espressif/esp32-camera #593: Camera init failed with error 0x105 (Help)

Petitelevelupe, 2023-11-01:
Any help pls?

E (10486) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105

Im trying to use the ESP32 Cam for a proyect so i need help whit this Error
Im using the examples from Arduino and got everything i need

pls help ;(

WangYuxin-esp, 2023-11-03:
Hello, we suggest that you first check your line connection according to the description [here](https://randomnerdtutorials.com/program-upload-code-esp32-cam/).

Oreobird, 2023-12-05:
Same issue happen to me, I use ESP32-S2-Kaluga-1 V1.3 developement Kit, its camera sensor is OV2640. Steps to reproduce the  issue:
(1) change the pins map in the example for Kaluga board as below：
```
// #define BOARD_WROVER_KIT 1
#define BOARD_ESP32_S2_KALUGA 1

#ifdef BOARD_ESP32_S2_KALUGA
#define CAM_PIN_PWDN -1
#define CAM_PIN_RESET -1
#define CAM_PIN_XCLK 1
#define CAM_PIN_SIOD 8
#define CAM_PIN_SIOC 7
#define CAM_PIN_D7 38
#define CAM_PIN_D6 21
#define CAM_PIN_D5 40
#define CAM_PIN_D4 39
#define CAM_PIN_D3 42
#define CAM_PIN_D2 41
#define CAM_PIN_D1 37
#define CAM_PIN_D0 36
#define CAM_PIN_VSYNC 2
#define CAM_PIN_HREF 3
#define CAM_PIN_PCLK 33
#endif
```
(2) build and flash, get the error info:
[image]

Funny thing is when I use ESP32-WROOM-32 module and connect with the same camera sensor，I can run my demo. It seems that I bought a broken ESP32-S2-Kaluga-1 V1.3 board? or any config I miss?

WangYuxin-esp, 2023-12-05:
Please check the level required for your camera module and the level provided by the board.

Oreobird, 2023-12-05:
The camera module voltage level is 3.3v and the board provides 3.34v. 
CAM_PIN_RESET is also 3.3v，may this affect sccb_prob ?

WangYuxin-esp, 2023-12-05:
This doesn't seem to affect. I think this is most likely a hardware issue. The pin configuration and hardware connection need to be completely correct. In addition, there are some [examples](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera) for reference here.

mthii, 2024-02-08:
This issues are the same here(ai thinker esp32-cam board with ov2640). This camera
does not have the camera rest pin(-1). That's why even the sample take_picture are intend to execute 'ov2640.c reset()' as follows. But it seems camera reset is intend to reset all the system to me. I don't know why.

```
101 static int reset(sensor_t *sensor)
102 {
103     int ret = 0;
104     WRITE_REG_OR_RETURN(BANK_SENSOR, COM7, COM7_SRST);
105     vTaskDelay(10 / portTICK_PERIOD_MS);
106     WRITE_REGS_OR_RETURN(ov2640_settings_cif);
107     return ret;
108 }
```

This function seems to reset whole system as follows,
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15556
load:0x40080400,len:4
0x40080400: _init at ??:?
```

```
I (00:00:07.895) main_task: Calling app_main()
I (00:00:07.900) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
D (928) intr_alloc: Connected src 22 to int 13 (cpu 0)
I (00:00:07.915) cam_hal: cam init ok
D (00:00:07.919) camera: Enabling XCLK output
D (00:00:07.924) ledc: Using clock source 4 (in slow mode), divisor: 0x200
D (00:00:07.930) ledc: In slow speed mode, global clk set: 4
D (00:00:07.936) ledc: LEDC_PWM CHANNEL 0|GPIO 00|Duty 0001|Time 0
D (00:00:07.942) camera: Initializing SCCB
I (00:00:07.946) sccb: pin_sda 26 pin_scl 27
I (00:00:07.951) sccb: sccb_i2c_port=1
D (968) intr_alloc: Connected src 50 to int 17 (cpu 0)
D (00:00:07.961) camera: Resetting camera by power down line
I (00:00:07.966) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (00:00:07.990) camera: Searching for camera address
I (00:00:08.001) camera: Detected camera at address=0x30
I (00:00:08.004) camera: Detected OV2640 camera
D (00:00:08.004) ov2640: OV2640 Attached
I (00:00:08.005) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
D (00:00:08.012) camera: Doing SW reset of sensor
'''

yoones1127, 2024-03-31:
If you use the ready-to-use code inside Ardavino, which is the same as Eczapala, the web server means that ESP32-cam must be connected to a network like a modem or WiFi that this pumpkin will find and change:

const char* ssid = "WiFi name";
const char* password = "password wifi";

closed 2024-06-06 by github-actions[bot]