espressif/esp32-camera #595: Adapting ESP32-cam-webserver example to use OV7725 image sensor

aranFish, 2023-11-04:
Hello,

I have an ESP32-CAM AI-Thinker module, with an OV2640 image sensor attached to its FFC connector.
I have successfully uploaded the [ESP32-CAM-webserver example code](https://github.com/easytarget/esp32-cam-webserver/tree/master) using both Arduino IDE and PlatformIO. It's working very well.

I would like to adapt this webserver example code to use the OV7725 image sensor. I have [this OV7725 image sensor module](https://www.uctronics.com/camera-modules/cmos-ov7725-camera-module-1-4-inch-0-3-megapixel-module.html), with an FFC connector and pinout that matches the pinout of the original OV2640 image sensor. I also have a short 3cm 24-way FFC cable to link the two modules together.

How can I use the device driver in the esp32-camera respository to do this? Any help you can offer would be appreciated.

Thank you.

A.

me-no-dev, 2023-11-08:
you connect the sensor that you want to the board, make sure that you have selected settings that it supports and that you have enabled the driver in menuconfig, then start the driver the same exact way. It will detect the sensor and do it's thing.

closed 2023-11-08 by me-no-dev

aranFish, 2023-11-09:
Thank you - I'll try this out. That sounds a lot simpler than I thought.

cheapohubby, 2024-05-19:
Did you ever get esp32camwebserver working with the OV7725?  I'm also trying to get it to work.  Initially ran into the PIXFORMAT_JPEG not being supported leading to a continuous reboot.  Changed to PIXFORMAT_RGB565, but still can't get any images out.  Went delving into the cam_hal.c code, and it seems like the psram_mode if false, and that seems forced by the CONFIG_IDF_TARGET_ESP32 being set.  I've verified that the esp32cam board has PSRAM (4MB).  Any suggestions?  Or maybe there's another example app that would allow me to at least pull some images out to see if the imager is worth the effort (trying to set up an IR camera for outdoors).

a26oss, 2025-10-20:
Hello, have you completed your project? Is it possible for you to release code for it? I've been trying to set up an outdoor night camera but i can't get the code to work at all. It would be much appreciated