espressif/esp32-camera #596: Easier setup for PlatformIO for ESP-IDF using idf_component.yml

nicklasb, 2023-11-05:
Hi, 

So I have been a bit annoyed by how complex this library is to use in PlatformIO and ESP-IDF, as mention [here](https://esp32.com/viewtopic.php?t=27293) it is quite the chore to get it building. So first, I went another way and cobbled the files together using CMake instead (my env is ESP32cam and it is a esp32 target): 

```CMake
FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/examples/*.*)
FILE(GLOB espcam_sources ${CMAKE_SOURCE_DIR}/.pio/libdeps/ESP32cam/esp32-camera/target/*.*)
FILE(GLOB_RECURSE espcam_sources_esp32 ${CMAKE_SOURCE_DIR}/.pio/libdeps/ESP32cam/esp32-camera/target/esp32/*.*)

idf_component_register(
    SRCS ${app_sources} ${espcam_sources} ${espcam_sources_esp32}
    INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/.pio/libdeps/ESP32cam/esp32-camera/target/jpeg_include 
        ${CMAKE_SOURCE_DIR}/.pio/libdeps/ESP32cam/esp32-camera/target/private_include
)
```
I certainly did not like that solution, as it is just everything else than portable. 

Instead, the easiest way for me was to be add an idf_component.yml-file and add esp32-camera as a simple dependency:
```YAML
version: "0.1.0"
description: "Testing esp32-camera"
dependencies:
  # Required IDF version
  idf: ">=4.1"
  esp32-camera:
    git: https://github.com/espressif/esp32-camera

```
and a 
```CMake

 REQUIRES esp32-cam
```
..to the CMakeLists.txt and it works? (or at least it builds)

So wouldn't the idf_component.yml path be much easier, or at least deserved to be mentioned, for most PlatformIO ESP-IDF users?

me-no-dev, 2023-11-08:
We do not control what and how is installed using PIO. What you ask about is documented for ESP-IDF. The component is standard structure and has no issues being used in ESP-IDF projects otherwise.

closed 2023-11-08 by me-no-dev

nicklasb, 2023-11-09:
> We do not control what and how is installed using PIO. What you ask about is documented for ESP-IDF. The component is standard structure and has no issues being used in ESP-IDF projects otherwise.

I am merely suggesting that there are two more ways of using the component with the ESP-IDF platform in PlatformIO.
And that the latter is not only easier than the one suggested in your documentation, but less "wonky", as you put it, and more aligned with the ESP-IDF component model as it uses the idf_component.yml path. 
Obviously, you don't have to agree.

me-no-dev, 2023-11-09:
I am a bit confused. Which documentation are you quoting? You opted to use the package manager, which is the default documented way going forward. Did you find some instructions elsewhere?

nicklasb, 2023-11-09:
> I am a bit confused. Which documentation are you quoting? You opted to use the package manager, which is the default documented way going forward. Did you find some instructions elsewhere?

The installation instructions on the repo first page (and below):
https://github.com/espressif/esp32-camera#platformio-libsubmodule-for-frameworkespidf

As according in the documentation above it, the package manager method only work with the Arduino platform.

igrr, 2023-11-09:
Just my 2c: since
```
dependencies:
  espressif/esp32-camera: "^2.0.5"
```
works in IDF build system (whether you invoke CMake from `idf.py` or from PlatformIO), it might be the easiest option to document that as a single installation method at this point?

There is a fair amount of text in the installation section of the project readme now, mentioning also submodules as an installation method. We could simplify this all to a fairly brief set of instructions based on idf_component.yml.

nicklasb, 2023-11-09:
> Just my 2c: since
>
> ```
> dependencies:
>   espressif/esp32-camera: "^2.0.5"
> ```
>
> works in IDF build system (whether you invoke CMake from `idf.py` or from PlatformIO), it might be the easiest option to [...]

That would be even better of course, just pointed to master since the documentation did that for ESP-IDF.

igrr, 2023-11-09:
Have opened https://github.com/espressif/esp32-camera/pull/597. @nicklasb Could you please sanity-check what I wrote from a perspective of a PlatformIO user?

nicklasb, 2023-11-09:
> Have opened #597. @nicklasb Could you please sanity-check what I wrote from a perspective of a PlatformIO user?

Tested using the dependency.  Worked. 
I would make PlatformIO bold or something, it is easy to miss that alternative in the text.

Thanks!

juliankrieger, 2024-04-10:
@nicklasb I am currently struggling with the same problem - just for a quick reference, using `idf_component.yml` and adding `REQUIRES` for my dependency in CMakeLists.txt -- for example for the main component -- in `idf_component_register` should just work?

nicklasb, 2024-04-10:
Yes, worked with having this dependency, haven't build it for a while but should still work. Not sure if you have to REQUIRE if it is in the idf_component.yml, actually. Note that you cannot have really old versions of ESP-IDF.

What is your problem?

juliankrieger, 2024-04-12:
No problem here, I was just wondering since the documentation mentions both ways, but I didn't see any mention of having to use "REQUIRES" for esp internal dependencies and idf_components.yml for external dependencies from the component library.