espressif/esp32-camera #598: ESP32 cam connected to Arduino UNO works on still images but does not work on video stream

eminahodzic1, 2023-11-09:
Hi,
So I have been working on a project for object detection. I am using ESP32 cam module AI Thinker and connecting it to my laptop using Arduino UNO. Everything works good but on the server it can only take still pictures but it does not work on live stream video, it is just a black screen. Does anyone know what is wrong? I don't think ist the hardware bcs everything else works, also it is connected with 5V of power and I don't think that is the problem either. Thankss in advance

me-no-dev, 2023-11-09:
Are you sure that you are running on Arduino UNO? Maybe you are using ESP32 Arduino to load up the web server example? Can you provide any pictures/code/terminal output of your setup?

eminahodzic1, 2023-11-09:
> Are you sure that you are running on Arduino UNO? Maybe you are using ESP32 Arduino to load up the web server example? Can you provide any pictures/code/terminal output of your setup?

Here is what i get in serial monitor:load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (631) esp_core_dump_flash: No core du� partition found!
E (632) esp_core_dump_flash: No core dump partition found!
E (425) psram: PSRAM ID read error: 0xffffffff

.........
WiFi connected
Camera Ready! Use 'http://172.20.21.58' to connect
Also, im just using one of the example codes in Arduino IDEm the WifiCamera one

me-no-dev, 2023-11-09:
in that case, as long as you are getting still images fine, video streaming should also work. Could it be your browser not being happy?

eminahodzic1, 2023-11-09:
> in that case, as long as you are getting still images fine, video streaming should also work. Could it be your browser not being happy?

I am using safari, I just tried chrome and there when i click to stream the video I get the picture again. So I guess thats something?

closed 2023-11-10 by eminahodzic1