espressif/esp32-camera #599: unable to include cam_hal.h

Abhay-2412, 2023-11-13:
Hello iam using arduino as component in esp-idf i have made some changes and wrote some own fucntions in cam_hal.c and when i try to access my functions by including cam_ha.h iam getting error

me-no-dev, 2023-11-13:
Sorry, but this is not at all enough information. You used arduino as component, yet you are posting in the camera repo about camera things. It's not at all clear:
- which ESP-IDF version
- which esp32-arduino version
- how did you add the camera component
- how does your main.cpp look like

Abhay-2412, 2023-11-13:
im posting in camera respo because because iam trying to use cam_hal.h fucntions in my source code

below is the file structure

```
project/
|-- main/
|   |-- CMakeLists.txt
|   |-- main.cpp
|   |-- Camera_Settings.cpp
|   |-- Camera_Settings.h
|   |-- DriverModules/
|   |   |-- BluetoothDriver/
|   |   |   |-- BLE_esp.cpp
|   |   |   |-- BLE_esp.h
|   |   |-- LEDdimmerDriver/
|   |   |   |-- PCA9352_LED_dimmer.cpp
|   |   |   |-- PCA9352_LED_dimmer.h
|   |   |-- HVDCDriver/
|   |   |   |-- Digipot_MCP4017.cpp
|   |   |   |-- Digipot_MCP4017.h
|   |   |   |-- HVDC_ADC_Settings.cpp
|   |   |   |-- HVDC_ADC_Settings.h
|   |   |-- BatteryDriver/
|   |       |-- BQ25601RTWR.cpp
|   |       |-- BQ25601RTWR.h
|   |-- Application/
|       |-- WifiDriver/
|           |-- WIFI_Client_Settings.cpp
|           |-- WIFI_Client_Settings.h
|-- components/
|   |-- espressif_esp32-camera/
|       |-- ...
|   |-- arduino/
|       |-- ...

```
iam using ESP-IDF version 4.4.5 and arduino of version 2.0.0

i added camera component as by creating idf_component.yml in main folder and added below lines
 ```
dependencies:
   espressif/esp32-camera: "^2.0.5"
```
but i made some changes in files so i moved above files to componet folder as u can see in file structure
my main.cpp look like as shown below
```

//file: main.cpp
#include "Arduino.h"

void setup(){
  Serial.begin(115200);

}

void loop(){
}
```
i have simple query how to use fucntions in cam_hal.c in main.cpp or other .cpp files

me-no-dev, 2023-11-13:
if you added the camera driver through `idf_component.yml` then it will be downloaded into `[project]/managed_components/espressif__esp32-camera`, but you might not want that, because you want to edit it.
Remove it from `idf_component.yml`, pull it into `components/esp32-camera` and then add it to `main/CMakeLists.txt` as `REQUIRES esp32-camera`.

Abhay-2412, 2023-11-13:
Hi @me-no-dev ,

Following the prescribed steps, a new issue has cropped up, throwing this error:

```
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/9NOV/GAZZELE_MINI/main/main.cpp:12:10: fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"
          ^~~~~~~~~~~
compilation terminated.
```
in all cpp files i got this error below is my cmake

```
idf_component_register(SRCS 
    "main.cpp" 
    "DriverModules/BluetoothDriver/BLE_esp.cpp"
    "Application/WifiDriver/WIFI_Client_Settings.cpp"
    "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.cpp"
    "Camera_Settings.cpp"
    "DriverModules/HVDCDriver/Digipot_MCP4017.cpp"
    "DriverModules/HVDCDriver/HVDC_ADC_Settings.cpp"
    "DriverModules/BatteryDriver/BQ25601RTWR.cpp"
    
    INCLUDE_DIRS
    "."
    "DriverModules/BluetoothDriver"
    "Application/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"

    REQUIRES
    esp32-camera
  
    )
```

me-no-dev, 2023-11-13:
Just add `arduino` after `esp32-camera` in CMakeLists.txt

Abhay-2412, 2023-11-13:
hello @me-no-dev 

I made changes in CMakeLists.txt as suggested
```

idf_component_register(SRCS 
    "main.cpp" 
    "DriverModules/BluetoothDriver/BLE_esp.cpp"
    "Application/WifiDriver/WIFI_Client_Settings.cpp"
    "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.cpp"
    "Camera_Settings.cpp"
    "DriverModules/HVDCDriver/Digipot_MCP4017.cpp"
    "DriverModules/HVDCDriver/HVDC_ADC_Settings.cpp"
    "DriverModules/BatteryDriver/BQ25601RTWR.cpp"
    
    INCLUDE_DIRS
    "."
    "DriverModules/BluetoothDriver"
    "Application/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"

    REQUIRES
    esp32-camera
    arduino
    
  
    )
```

but still getting same error as before

```
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/9NOV/GAZZELE_MINI/main/main.cpp:41:10: fatal error: cam_hal.h: No such file or directory
 #include "cam_hal.h"
          ^~~~~~~~~~~
```

me-no-dev, 2023-11-13:
That now probably has to do with the fact that `cam_hal.h` is PRIVATE include in `esp32-camera`. You would need to adjust the camera CMakeLists.txt too to make it not private

Abhay-2412, 2023-11-13:
Hello thanks for the response!

Can you kindly mention steps on how to remove them from private i did'nt make any changes in camera cmake yet

me-no-dev, 2023-11-13:
the easiest way would be to just move the header from `components/esp32-camera/driver/private_include` to `components/esp32-camera/driver/include`

Abhay-2412, 2023-11-14:
yep error resolved thanks @me-no-dev appreciate for your help

closed 2023-11-14 by Abhay-2412