espressif/esp32-camera #600: Guru Meditation Error: Core  1 panic'ed (StoreProhibited

Abhay-2412, 2023-11-17:
Hello iam using DFrobot firebeetle 2 esp32s3 with ov5640 iam trying to build my code in esp idf version 4.4.5 with arduino as component of version 2.0.6 iam getting below error please help me in trouble shooting it and iam happy to share any additional information

```
I (1835218) s3 ll_cam: node_size: 2400, nodes_per_line: 1, lines_per_node: 1
I (1835219) s3 ll_cam: dma_half_buffer_min:  2400, dma_half_buffer: 12000, lines_per_half_buffer:  5, dma_buffer_size: 24000
I (1835227) cam_hal: Allocating 1920000 Byte frame buffer in PSRAM
I (1835234) cam_hal: cam config ok
I (1843076) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3
I (1843076) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 12, dma_buffer_size: 30720
I (1843085) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (1843092) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (1843098) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (1843105) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (1843112) cam_hal: cam config ok
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40056f60  PS      : 0x00060530  A0      : 0x8200a00c  A1      : 0x3fcebd70
A2      : 0x00000000  A3      : 0x3d8009a4  A4      : 0x0004b000  A5      : 0x00000000  
A6      : 0x873f9426  A7      : 0x7f189535  A8      : 0x00000000  A9      : 0x3fcebd20
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3fcecc24  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00004aff

Backtrace: 0x40056f5d:0x3fcebd70 |<-CORRUPTED
```

Abhay-2412, 2023-11-20:
atleast a suggestion on how to troubleshoot it will be appreciated

me-no-dev, 2023-11-20:
there are some strange config lines in the log. What is your configuration for the camera driver? Some init code could also help.

Abhay-2412, 2023-11-21:
Hello thanks for the reply still struggling with same issue my requirement is i want to capture one pic from one resolution(1200x800) and capture rest of the pics with other resolution(640x240), i have acheived my requirement in arduino ide version 2.0.6 thats the reason iam using arduino as a component in esp idf but strangely iam only able to receive one picture(1200x800) and after that my code is crashing 

first i init my camera with below function

```
esp_err_t initCameraSettings(void)
{
  //Camera configuration
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 24000000;
  config.frame_size = FRAMESIZE_HD;
  config.pixel_format = PIXFORMAT_YUV422; // for streaming
  //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 4;

  // Camera init
  esp_err_t err = esp_camera_init(&config);
  sensor_t * s = esp_camera_sensor_get();
  //Check for camera PID and apply settings
  if (s->id.PID == OV5640_PID)
  {
    Serial.println("OV5640 camera is connected");
    s->set_brightness(s,0);
    s->set_contrast(s,0);
    s->set_saturation(s,0);
    s->set_sharpness(s,0);
    s->set_denoise(s,0);
    s->set_ae_level(s,-3);
//    s->set_gainceiling(s, (gainceiling_t)0);
    s->set_special_effect(s,0);
    s->set_whitebal(s,1);
    s->set_exposure_ctrl(s,1);
    s->set_res_raw(s, 680, 560, 1880, 1360, 0, 0, 5120, 1920, 1200, 800, false, false);
  }
  return err;
}
```
above function i wrote it in seprate cpp file i call above function in `main.cpp`

 ```
 //  // Camera initilization
  esp_err_t error = initCameraSettings();
  if (error != ESP_OK)
  {
    Serial.printf("Camera init failed with error 0x%x", error);
  }
  cam_stop();
```
after that i have writen code in such a way that first picture is of 1200x800 resolution rest of all pictures is of HVGA resolution

[... 103 lines cut ...]

  size_t frame_buffer_queue_len = cam_obj->frame_cnt;
  if (config->grab_mode == CAMERA_GRAB_LATEST && cam_obj->frame_cnt > 1) {
    frame_buffer_queue_len = cam_obj->frame_cnt - 1;
  }
  cam_obj->frame_buffer_queue = xQueueCreate(frame_buffer_queue_len, sizeof(camera_fb_t*));
  CAM_CHECK_GOTO(cam_obj->frame_buffer_queue != NULL, "frame_buffer_queue create failed", err);

// #if CONFIG_CAMERA_CORE0
  xTaskCreatePinnedToCore(cam_task, "cam_task", 2048, NULL, configMAX_PRIORITIES - 1, &cam_obj->task_handle, 0);
// #elif CONFIG_CAMERA_CORE1
  // xTaskCreatePinnedToCore(cam_task, "cam_task", 2048, NULL, configMAX_PRIORITIES - 2, &cam_obj->task_handle, 1);
// #else
  // xTaskCreate(cam_task, "cam_task", 2048, NULL, configMAX_PRIORITIES - 2, &cam_obj->task_handle);
// #endif

  ESP_LOGI(TAG, "cam config ok");
  return ESP_OK;

err:
  cam_deinit();
  return ESP_FAIL;
}

```

closed 2024-01-29 by github-actions[bot]