espressif/esp32-camera #602: Any support for OV5647?

stellna, 2023-11-21:
Hello, I'm recently practicing using esp32. I have created a task using esp32 ov2640 camera, but it resolution is so low. So, I just bought an OV5647 camera to resolve this problem. But I need to change this line of code to match OV5647. Any idea or help will be appreciate. Thank you.

static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,  //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QVGA,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 10,  //0-63 lower number means higher quality
    .fb_count = 1       //if more than one, i2s runs in continuous mode. Use only with JPEG
};

static esp_err_t init_camera(){
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }

    return ESP_OK;
}

For more precise information, Here is my github link for the project i'm working on.
https://github.com/stellna/IOT-camera-learning

me-no-dev, 2023-11-22:
We do not support currently that sensor and generally the speed depends as much from the sensor as it does from what the chip can really achieve. OV5640 provides 5MP images, but is also very slow at high resolutions.

What are your project requirements?

stellna, 2023-11-22:
Well, There is no specific requirements in my project.
It is just that currently, i'm using esp-idf human_face_detect DL header files. But it just keep showing human detected when there is no human in the picture. So, I thought that it maybe just because of the resolution.

closed 2024-01-30 by github-actions[bot]

luistorres00, 2024-08-26:
what are your recommendation for a full hd resolution for streamming with the esp32 cam?