espressif/esp32-camera #605: LCD_CAM interrupt allocation of camera failed

eclps123, 2023-11-27:
同时使用i80bus和摄像头模块，会出现 LCD_CAM interrupt allocation of camera failed。可能是中断配置方面的问题，但我不知道如何解决它

`I (27) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (27) boot: compile time Nov 27 2023 17:19:52
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 phy_init         RF data          01 01 00009000 00001000
I (72) boot:  1 factory          factory app      00 00 00010000 001f4000
I (80) boot:  2 nvs              WiFi data        01 02 00204000 00004000
I (87) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=27968h (162152) map
I (129) esp_image: segment 1: paddr=00037990 vaddr=3fc94b00 size=0408ch ( 16524) load
I (133) esp_image: segment 2: paddr=0003ba24 vaddr=40374000 size=045f4h ( 17908) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=50ee0h (331488) map
I (203) esp_image: segment 4: paddr=00090f08 vaddr=403785f4 size=0c4bch ( 50364) load
I (221) boot: Loaded app from partition at offset 0x10000
I (222) boot: Disabling RNG early entropy source...
I (233) cpu_start: Multicore app
I (233) octal_psram: vendor id    : 0x0d (AP)
I (234) octal_psram: dev id       : 0x02 (generation 3)
I (237) octal_psram: density      : 0x03 (64 Mbit)
I (242) octal_psram: good-die     : 0x01 (Pass)
I (247) octal_psram: Latency      : 0x01 (Fixed)
I (253) octal_psram: VCC          : 0x01 (3V)
I (258) octal_psram: SRF          : 0x01 (Fast Refresh)
I (264) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (269) octal_psram: BurstLen     : 0x01 (32 Byte)
I (275) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (281) octal_psram: DriveStrength: 0x00 (1/1)
I (287) MSPI Timing: PSRAM timing tuning index: 5
I (292) esp_psram: Found 8MB PSRAM device
I (296) esp_psram: Speed: 80MHz
I (300) cpu_start: Pro cpu up.
I (304) cpu_start: Starting app cpu, entry point is 0x40375590
0x40375590: call_start_cpu1 at C:/Espressif/511/frameworks/esp-idf-v5.1.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (763) esp_psram: SPI SRAM memory test OK
I (772) cpu_start: Pro cpu start user code
I (772) cpu_start: cpu freq: 160000000 Hz
I (772) cpu_start: Application information:
I (775) cpu_start: Project name:     i80_controller
I (780) cpu_start: App version:      1
I (785) cpu_start: Compile time:     Nov 27 2023 18:47:58
I (791) cpu_start: ELF file SHA256:  e5a9ea10df6bba9a...
I (797) cpu_start: ESP-IDF:          v5.1.1-dirty
I (802) cpu_start: Min chip rev:     v0.0
I (807) cpu_start: Max chip rev:     v0.99 
I (812) cpu_start: Chip rev:         v0.2
I (817) heap_init: Initializing. RAM available for dynamic allocation:
I (824) heap_init: At 3FCA1968 len 00047DA8 (287 KiB): DRAM
I (830) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (837) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (843) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (849) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (857) spi_flash: detected chip: generic
I (861) spi_flash: flash io: dio
I (865) sleep: Configure to isolate all GPIO pins in sleep state
I (872) sleep: Enable automatic switching of GPIO sleep configuration
I (879) app_start: Starting scheduler on CPU0
I (884) app_start: Starting scheduler on CPU1
I (884) main_task: Started on CPU0
I (894) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (904) main_task: Calling app_main()
I (904) s3 ll_cam: DMA Channel=4
I (914) cam_hal: cam init ok
I (914) sccb: pin_sda 9 pin_scl 46
I (914) sccb: sccb_i2c_port=0
I (924) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (934) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (944) example: Initialize Intel 8080 bus
I (944) example: Install LCD driver of nt35510
I (954) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (954) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (954) example: Display camera
I (994) camera: Detected camera at address=0x21
I (994) camera: Detected OV7725 camera
I (994) camera: Camera PID=0x77 VER=0x21 MIDL=0x7f MIDH=0xa2
I (1074) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (1074) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (1074) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1084) example: Turn on LCD backlight
I (1084) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (1094) example: Initialize LVGL library
E (1104) s3 ll_cam: LCD_CAM interrupt allocation of camera failed
I (1104) example: 551
E (1114) cam_hal: cam_config(401): cam intr alloc failed
I (1114) example: 557
E (1124) camera: Camera config failed with error 0xffffffff
I (1124) example: buf1@0x3c0b6380, buf2@0x3c0dd4a0
E (1124) example: Camera Init Failed
I (1134) example: Register display driver to LVGL
I (1134) example: camera init failed!!!!
I (1144) example: Install LVGL tick timer
I (1154) example: Display LVGL animation`

liruya, 2023-12-28:
同问, 我也遇到了同样的问题. ESP32S3 LCD用i80接口或者RGB接口, 同时使用Camera, 就会出现这种情况.
如果先初始化LCD, 则Camera初始化失败, 如果先初始化Camera, 则LCD初始化失败.
如果LCD用SPI接口, 则没有问题.
定位到  esp_intr_alloc_intrstatus -> get_available_int 这里, 中断配置失败. LCD和Camera的中断配置出现冲突, 不知道怎么解决.

liruya, 2024-01-02:
LCD和Camera初始化冲突找到解决办法, 确实是配置中断的冲突, LCD和Camera中断用同一个中断源, 导致分配中断失败.
解决办法是LCD中断和Camera中断分配到不同的CPU. 比如LCD在CPU0 任务里初始化, Camera在CPU1任务里初始化.
另外发现必须先初始化Camera再初始化LCD, 否则还是有异常, 这个没发现什么原因.

closed 2024-03-09 by github-actions[bot]