espressif/esp32-camera #608: Is there a problem with my distribution of these GPIO pins?

yjmthu, 2023-12-08:
Hello, I am using ESP32-PICO-D4 and OV2640.
The following code is the pin definition I am using.
```cpp
#define PWDN_GPIO_NUM -1
#define RESET_GPIO_NUM 2
#define XCLK_GPIO_NUM 14
#define SIOD_GPIO_NUM 4
#define SIOC_GPIO_NUM 0

#define Y2_GPIO_NUM 35
#define Y3_GPIO_NUM 39
#define Y4_GPIO_NUM 38
#define Y5_GPIO_NUM 34
#define Y6_GPIO_NUM 33
#define Y7_GPIO_NUM 26
#define Y8_GPIO_NUM 27
#define Y9_GPIO_NUM 12
#define VSYNC_GPIO_NUM 15
#define HREF_GPIO_NUM 13
#define PCLK_GPIO_NUM 25
```
This is the initialization function for my camera.
```cpp
void CamSetup() {
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG; 
  
  if(psramFound()) {
    config.frame_size = FRAMESIZE_HVGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    Serial.println("couldn't find psram for camera!");
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }
  
  // Camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }
}
```
The problem is that when I execute the `esp_camera_fb_get` function, it always returns a null pointer.
Is there something wrong with my pin definition?

me-no-dev, 2023-12-18:
do not use IO 0 for such digital signals. Often a capacitor is added to debounce the "BOOT" button and that interferes with high frequency signals, like I2C SCL. Probably the driver can not find the camera.

yjmthu, 2023-12-19:
Yes, there can be problems with using GPIO0. But I solved this problem by removing the GPIO2's connection and defining it as -1. This may be due to the capacitance or pull-up at io2.

closed 2023-12-19 by yjmthu