espressif/esp32-camera #611: Do not use abbreviations in logging

nicklasb, 2023-12-17:
Thanks for a nice library!

However, it would be even nicer with a hopefully simple change; remove abbreviations in logging. 
There is no need leaving users guessing with "FB-OVF" or "FBQ-SND". 
The bytes saved are quite thoroughly offset by the clarity and ease of understanding of the error messages.

me-no-dev, 2023-12-18:
The only reason why they are abbreviated is to be able to print them as fast as possible, because they happen in ISR that might get executed very often. Longer messages cause other issues. We should document them though!

nicklasb, 2023-12-18:
Ok, got it. 
But documentation (which wouldn't hur regardless) sounds good enough, and a short note of this somewhere early in the docs would be great. Closing!

closed 2023-12-18 by nicklasb