espressif/esp32-camera #612: Calling set_framesize() with PIXFORMAT_GRAYSCALE yields a familiar failure mode upon capture

MikeSnyder360, 2023-12-20:
I'm seeing the same issues as this: https://github.com/espressif/esp32-camera/issues/514

- using PIXFORMAT_GRAYSCALE, trying to call set_framesize() at runtime 
  - the configured buffer size mismatch causes the FB-SIZE warning to be printed
  - https://github.com/espressif/esp32-camera/blob/master/driver/cam_hal.c#L195
  - a stack overflow occurs

to replicate -
- boot with FRAMESIZE_QVGA
- call set_framesize(s,FRAMESIZE_240X240)
- capture an image

I changed to PIXFORMAT_JPEG, the problem immediately goes away. 

The problem is I need raw grayscale for my application. 
I don't see a direct JPG -> grayscale API. 
I can convert from JPG to RGB565, and then manually convert that to grayscale. 
That's a bit much for a workaround, my task is already quite slow.

oldmud0, 2024-01-04:
Have you tried a manual delay between setting the frame size and grabbing a frame from the frame buffer? Obviously this is not ideal but it may work around the timeout issue.

PatrikRindlisbacher, 2024-01-05:
I'm struggling with the same problem. M5 Stack Timer Camera X

me-no-dev, 2024-01-08:
@MikeSnyder360 the driver currently does not support changing the resolution, when not in JPEG mode. It is planned along with a few more things. This is due to the fact that the whole peripheral and DMA need to be reconfigured. In JPEG mode, only setting to lower resolution/quality than the one in config is possible, because the same buffer can be used and nothing needs to be reconfigured.

PatrikRindlisbacher, 2024-01-08:
I thought my problem had the same cause. 
But because I don't switch s, the cause seems to be something else. 
Do I have to open a new issue?

AgainPsychoX, 2024-02-25:
While **current version of library doesn't support changing pixel format nor frame size on the fly**, I have some **success with re-initializing** the camera. You need to use `esp_camera_deinit()` and `esp_camera_init(&camera_config)` with desired `pixel_format`/`frame_size` there.

AxelLin, 2024-02-27:
@me-no-dev 

Test with v5.1.2-695-gd922d4178f + esp32-camera 2.0.7.
I didn't change resolution at runtime.
I use with below settings in my camera_config.

    .pixel_format = PIXFORMAT_GRAYSCALE,    
    .frame_size = FRAMESIZE_SVGA,
    .jpeg_quality = 6,
    .fb_count = 1,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,

When test on ESP32, it works with PIXFORMAT_GRAYSCALE.
But when test on ESP32S3, I got "cam_hal: EV-VSYNC-OVF", see below log:

I (6342) s3 ll_cam: DMA Channel=4
I (6342) cam_hal: cam init ok
I (6342) sccb: pin_sda 18 pin_scl 8
I (6343) sccb: sccb_i2c_port=1
I (6357) camera: Detected camera at address=0x30
I (6360) camera: Detected OV2640 camera
I (6360) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (6443) s3 ll_cam: node_size: 3200, nodes_per_line: 1, lines_per_node: 2
I (6444) s3 ll_cam: dma_half_buffer_min:  3200, dma_half_buffer: 16000, lines_per_half_buffer: 10, dma_buffer_size: 32000
I (6452) cam_hal: buffer_size: 32000, half_buffer_size: 16000, node_buffer_size: 3200, node_cnt: 10, total_cnt: 60
I (6463) cam_hal: Allocating 480000 Byte frame buffer in PSRAM
I (6470) cam_hal: cam config ok
I (6473) ov2640: Set PLL: clk_2x: 1, clk_div: 7, pclk_auto: 1, pclk_div: 8
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
W (10553) cam_hal: Failed to get the frame on time!

Also note, It works for both ESP32 and ESP32S3 if using PIXFORMAT_JPEG.

closed 2024-05-05 by github-actions[bot]