espressif/esp32-camera #614: s3 ll_cam: DMA interrupt allocation of camera failed

pandiannano, 2024-01-02:
I used the component library of esp32-camera(v2.0.6) on esp32s3(M5CoreS3) and added esp idf v5.0.2 with micropython v1.21.0. it is our projects

Problems:
DMA interrupt allocation of camera failed will occur. There might be a problem with the DMA interrupt configuration but I don't know how to fix it.

D (19703) spi_master: device5 locked the bus
D (19735) spi_master: device5 release bus
D (19736) spi_master: device5 locked the bus
D (19768) spi_master: device5 release bus
I (19768) s3 ll_cam: DMA Channel=4
I (19769) cam_hal: cam init ok
D (19769) camera: Initializing SCCB
I (19772) sccb: pin_sda 12 pin_scl 11
I (19777) sccb: sccb_i2c_port=1
D (19781) intr_alloc: Connected src 43 to int 23 (cpu 0)
D (19786) camera: Searching for camera address
I (19800) camera: Detected camera at address=0x21
I (19801) ov7725: Mismatch PID=0xe8
I (19802) ov7760: Mismatch PID=0xe8
I (19805) gc032a: Mismatch PID=0xf302
I (19809) camera: Detected GC0308 camera
D (19813) gc0308: GC0308 Attached
I (19816) camera: Camera PID=0x9b VER=0x00 MIDL=0x00 MIDH=0x00
D (19823) camera: Doing SW reset of sensor
D (20011) gc0308: Camera defaults loaded
I (20091) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (20091) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (20099) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (20110) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (20117) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
E (20124) s3 ll_cam: DMA interrupt allocation of camera failed
E (20130) cam_hal: cam_config(401): cam intr alloc failed
E (20136) camera: Camera config failed with error 0xffffffff
E (20143) m5camera: Camera init Failed
E (20148) m5camera: Camera capture Failed
E (20152) m5camera: Camera capture Failed
E (20157) m5camera: Camera capture Failed
E (20161) m5camera: Camera capture Failed
E (20166) m5camera: Camera capture Failed
E (20171) m5camera: Camera capture Failed

liqijian101, 2024-01-07:
I have encountered the same problem. Is there a solution？ Thaks

me-no-dev, 2024-01-08:
@liqijian101 also IDF 5.0.x?

liqijian101, 2024-01-08:
IDF 5.1.1.
And I use an LCD screen, which uses an SPI driver.

liqijian101, 2024-01-08:
@me-no-dev

me-no-dev, 2024-01-09:
I will try this out, though we use IDF 5.1 in Arduino and all is fine. Could any one of you send me their sdkconfig? I would like to make sure that all settings are the same.

liqijian101, 2024-01-09:
[sdkconfig.zip](https://github.com/espressif/esp32-camera/files/13872887/sdkconfig.zip)

@me-no-dev  thanks

liqijian101, 2024-01-14:
With IDF v5.1.2, there is no problem  @me-no-dev

ConnorDuvalS, 2024-03-05:
Encountering this issue as well. I can confirm it is caused by the call to `get_available_int` in `esp_intr_alloc_intrstatus` failing in the ESP32-S3 version of `ll_cam_init_isr`. I can replicate on ESP-IDFv5.1.3 and the latest `release/v5.1` push, `d22c95a3a8f9ffa760567648531c870afc16ab87`, but not with ESP-IDFv5.1.2 (at the `v5.1.2` tag), likely a degradation case on the framework.

liqijian101, 2024-03-21:
Do you have a solution？ @ConnorDuvalS

liqijian101, 2024-03-21:
How to configure sdkconfig to prevent this issue in ESP32S3 @me-no-dev

AxelLin, 2024-05-22:
@pandiannano @liqijian101 
Do you still hit this issue with latest esp32-camera(v2.0.8)?

tsy2001, 2024-06-11:
I also encountered the same error：
[image]

closed 2024-08-18 by github-actions[bot]