espressif/esp32-camera #620: esp_camera_fb_get() fails on ESP32S3 after device joins an AP

AxelLin, 2024-01-18:
Test on esp-idf master and  5.2 branch.

My configuration is:

```
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,
    .frame_size = FRAMESIZE_SVGA,
    .jpeg_quality = 12,
    .fb_count = 1,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
};
```

test code:
```
while (1) {
  ESP_LOGI(TAG, "Taking picture...");
  camera_fb_t *pic = esp_camera_fb_get();

  // use pic->buf to access the image
  if (pic) {
  ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
    esp_camera_fb_return(pic);
  }

  vTaskDelay(pdMS_TO_TICKS(5000));    /* small delay to print debug messages */
}
```

If the device does not join any access point, just execute below test code works on both ESP32 and ESP32S3.
However, if let the device join an access point then executing above test code does not work well on ESP32S3 (tested on ESP32S3-WROOM, FREENOVE).
(It still works for ESP32).

In below log, I call esp_camera_fb_get() 3 times before join AP, and it works.
But after joined AP, esp_camera_fb_get() only works for 1 time.

I (1726) s3 ll_cam: DMA Channel=4
I (1728) cam_hal: cam init ok
I (1731) sccb: pin_sda 4 pin_scl 5
I (1735) sccb: sccb_i2c_port=1
I (1749) camera: Detected camera at address=0x30
I (1752) camera: Detected OV2640 camera
I (1752) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1831) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 93
I (1832) cam_hal: Allocating 96000 Byte frame buffer in PSRAM
I (1838) cam_hal: cam config ok
I (1842) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
I (1925) test_cam: Taking picture...
I (1987) test_cam: Picture taken! Its size was: 18258 bytes
I (6987) test_cam: Taking picture...
I (6987) test_cam: Picture taken! Its size was: 18147 bytes
I (11987) test_cam: Taking picture...
I (11987) test_cam: Picture taken! Its size was: 17425 bytes
I (16989) NETWORK: WIFI listen_interval=0
I (16990) net_utils: Joining AP: IGS-MESH
I (16990) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (17039) wifi:mode : sta (34:85:18:6e:fa:7c)
I (17040) wifi:enable tsf
I (17041) net_utils: STA started
I (19450) wifi:new:<8,2>, old:<1,0>, ap:<255,255>, sta:<8,2>, prof:1
I (19810) wifi:state: init -> auth (b0)
I (19815) wifi:state: auth -> init (8a0)
I (19815) wifi:new:<8,0>, old:<8,2>, ap:<255,255>, sta:<8,2>, prof:1
I (19817) wifi:new:<8,2>, old:<8,0>, ap:<255,255>, sta:<8,2>, prof:1
I (19820) wifi:state: init -> auth (b0)
I (19841) wifi:state: auth -> assoc (0)
I (19848) wifi:state: assoc -> run (10)
I (19873) wifi:connected with IGS-MESH, aid = 8, channel 8, 40D, bssid = d8:47:32:7c:37:ae
I (19873) wifi:security: WPA2-PSK, phy: bgn, rssi: -25
I (19875) wifi:pm start, type: 1

I (19878) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (19886) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (19896) net_utils: STA connected
I (19901) wifi:<ba-add>idx:0 (ifx:0, d8:47:32:7c:37:ae), tid:6, ssn:2, winSize:64
I (19941) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (20896) esp_netif_handlers: sta ip: 192.168.1.128, mask: 255.255.255.0, gw: 192.168.1.1
I (25938) test_cam: Taking picture...
I (25938) test_cam: Picture taken! Its size was: 17348 bytes
I (30938) test_cam: Taking picture...
W (34938) cam_hal: Failed to get the frame on time!
I (39938) test_cam: Taking picture...
W (43938) cam_hal: Failed to get the frame on time!
I (48938) test_cam: Taking picture...
W (52938) cam_hal: Failed to get the frame on time!

NOTE: If set fb_count = 2, esp_camera_fb_get() will success for 2 times (if calling it after wifi connected).

AxelLin, 2024-01-22:
I just notice it works if build with v5.1.2-622-g47b44dfa71, so this looks like a regression in esp-idf.
I'll close this one and report it to esp-idf.

me-no-dev, 2024-01-22:
@AxelLin please provide a minimal (but complete) project example that I can compile here and reproduce. Also please add your sdkconfig too (needed to confirm all options are the same)

me-no-dev, 2024-01-22:
@AxelLin can you try the workaround [here](https://github.com/espressif/esp32-camera/pull/625)?

AxelLin, 2024-01-23:
> @AxelLin can you try the workaround [here](https://github.com/espressif/esp32-camera/pull/625)?

Hi @me-no-dev 
Above workaround works for both master and v5.2 branch.
Thanks.

closed 2024-01-23 by me-no-dev