espressif/esp32-camera #626: Cannot find devie's AP after init camera on ESP32S3

AxelLin, 2024-01-23:
Cannot find devie's AP after init camera on ESP32S3

Test on esp-idf master, 5.2 branch and 5.1 branch.

If I call esp_camera_init() before starting AP.
The console log does not show any error but I usually cannot find the device's AP.
In most of my test, I just cannot find the device's AP.
Sometimes, I can find the device's AP but failed to join it:
```
I (43253) wifi:removing station <20:1e:88:5e:2b:3f> after unsuccessful auth/assoc, AID = 0
W (43254) wifi:rm mis
I (43254) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
```

However, if I comment out the esp_camera_init() everything looks fine.

I also tested on ESP32 and ESP32 works, only ESP32S3 fails.
But on ESP32 the wifi is very slow after init camera.

me-no-dev, 2024-01-23:
I would guess that this is connected to the previous issue, though I do not see a workaround from our end for this.

@igrr would someone from IDF/WiFi team look into this?

me-no-dev, 2024-01-23:
@AxelLin can you try to run the camera on the other core? Let's see if maybe WiFi and Camera tasks starve each other

AxelLin, 2024-01-23:
> @AxelLin can you try to run the camera on the other core? Let's see if maybe WiFi and Camera tasks starve each other

I tried "CORE0", "CORE1" and "NO_AFFINITY" on v5.1.2-622-g47b44dfa71, the same issue.

igrr, 2024-01-23:
@AxelLin could you please add some details about the hardware (board) you are using?

AxelLin, 2024-01-23:
I tested on "Freenote ESP32-S3-WROOM Board".

AxelLin, 2024-01-24:
I did another test on S3:

1. Start AP first
   I can find the AP, and I can join the AP.
2. Delay 30 sec
   It looks fine before calling esp_camera_init()
3. Call esp_camera_init()
   After calling esp_camera_init() the AP disappear and I got below log immediately:
I (35486) wifi:station: 20:1e:88:5e:2b:3f leave, AID = 1, bss_flags is 658531, bss:0x3fcda26c
I (35487) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
I (35490) wifi:<ba-del>idx:2, tid:0

leeebo, 2024-01-24:
Hi @AxelLin I did the same test, `call esp_camera_init() before starting AP.` with `esp32s3-eye` hardware and ESP-IDF latest `release/v5.2`, both Camera and Wi-Fi work well for me. Would you please double check your hardware, whether there is a power supply or antenna problem ?

Xiehanxin, 2024-01-24:
hi @AxelLin from the log we can see 
_D (38797) wifi:ap recv deauth/disassoc from sta 20:1e:88:5e:2b:3f, reason:980_
it shows that the STA send the deauth/disassoc to AP actively. and the number 980 is got from the deauth/disassoc field.
the real reason might need a wireless package.

AxelLin, 2024-01-24:
>  Would you please double check your hardware, whether there is a power supply or antenna problem ?

@leeebo 
It works with STA + camera (v2.0.7) , so I think the power supply and the onboard antenna is fine.
(I also try power by adapter or power-bank, the same symptom).

Just wondering if you have a chance to test this board: https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board

AxelLin, 2024-01-24:
> hi @AxelLin from the log we can see _D (38797) wifi:ap recv deauth/disassoc from sta 20:1e:88:5e:2b:3f, reason:980_ it shows that the STA send the deauth/disassoc to AP actively. 

@Xiehanxin 

You seems imply the issue is beause the *sta* actively deauth/disassoc to AP.
But this symptom does not happen when camera is not inited.

Attached another detail log of verbose build:
I don't find any error message, only found below warning messages before deauth:
(ps. I don't observe below warngins if camera is not inited)
W (19059) wifi:[ADDBA]rx delba, code:37, delete tid:0
I (19060) wifi:<ba-del>idx:2, tid:0
W (19060) wifi:[ADDBA]rx delba, code:37, delete tid:0

Then the device got
wifi:ap recv deauth/disassoc from sta 20:1e:88:5e:2b:3f, reason:58

[s3_esp32cam_ap.log](https://github.com/espressif/esp32-camera/files/14033719/s3_esp32cam_ap.log)

AxelLin, 2024-01-26:
Hi @me-no-dev 

FYI, https://github.com/espressif/esp32-camera/pull/629 does not help for this issue.
I tried both LCD_CAM_ISR_IRAM_SAFE=y and LCD_CAM_ISR_IRAM_SAFE=n, no difference.
The AP still disappear after init camera.
Thanks anyway.

me-no-dev, 2024-01-26:
@AxelLin that would mean that the issue is not in GDMA or IRAM ISR. It does fix the previous (STA) issue, so you should not feel a delay when WiFi starts.

AxelLin, 2024-02-22:
I cannot reproduce this issue now after change test location.

closed 2024-02-22 by AxelLin