espressif/esp32-camera #627: Esp32-cam photos are very very bright - deep sleep problem ?

feiticeir0, 2024-01-23:
I'm experiencing weird problems with the ESP32-CAM photos when coming from **deep sleep.** 

The first two to three photos came well, but after that, they start getting white until I can't see anything.  This only happens outside and in daylight and only with **deep sleep.** 

The system is configure with a PIR sensor that, if detects activity, it wakes up from deep sleep, takes a picture, sends it to telegram and enters the deep sleep state again. 
After a while, they just keep getting whiter... 
Here's the first one while I was setting the ESP32-Cam:
[image]

After 3 more photos:
[image]
[image]
[image]
[image]

The thing is, if I try the webserver stream example and go outside, the image is always good.  I access the ESP32-CAM and see the stream and nothing like this happens. 

I've even tried in the code to configure the camera with the same default parameters that are defined in the webserver stream and it starts to happen the same thing.

Here's my camera sensor settings:

`  sensor_t * s = esp_camera_sensor_get();
  s->set_framesize(s, FRAMESIZE_QVGA);  // UXGA|SXGA|XGA|SVGA|VGA|CIF|QVGA|HQVGA|QQVGA
  //brightness
  s->set_brightness(s, 0);     // -2 to 2
  s->set_contrast(s, 0);       // -2 to 2
  s->set_saturation(s, 0);     // -2 to 2
  s->set_special_effect(s, 0); // 0 to 6 (0 - No Effect, 1 - Negative, 2 - Grayscale, 3 - Red Tint, 4 - Green Tint, 5 - Blue Tint, 6 - Sepia)
  s->set_whitebal(s, 1);       // 0 = disable , 1 = enable
  s->set_awb_gain(s, 1);       // 0 = disable , 1 = enable
  s->set_wb_mode(s, 0);        // 0 to 4 - if awb_gain enabled (0 - Auto, 1 - Sunny, 2 - Cloudy, 3 - Office, 4 - Home)
  s->set_exposure_ctrl(s, 1);  // 0 = disable , 1 = enable
  s->set_aec2(s, 0);           // 0 = disable , 1 = enable
  s->set_ae_level(s, 0);       // -2 to 2
  s->set_aec_value(s, 300);    // 0 to 1200
  s->set_gain_ctrl(s, 1);      // 0 = disable , 1 = enable
  s->set_agc_gain(s, 0);       // 0 to 30
  s->set_gainceiling(s, (gainceiling_t)0);  // 0 to 6
  s->set_bpc(s, 0);            // 0 = disable , 1 = enable
  s->set_wpc(s, 1);            // 0 = disable , 1 = enable
  s->set_raw_gma(s, 1);        // 0 = disable , 1 = enable
  s->set_lenc(s, 1);           // 0 = disable , 1 = enable
  s->set_hmirror(s, 0);        // 0 = disable , 1 = enable
  s->set_vflip(s, 0);          // 0 = disable , 1 = enable
  s->set_dcw(s, 1);            // 0 = disable , 1 = enable
  s->set_colorbar(s, 0);       // 0 = disable , 1 = enable`

With or without this settings, the end result is always the same - after a while, full white images. 
I've already tried all the ESP32-Cam that I have (5 of them) and they all do the same.

Here's the full code.

`#include <Arduino.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "esp_camera.h"
#include <UniversalTelegramBot.h>
#include <ArduinoJson.h>

const char* ssid = "";
const char* password = "";

// Initialize Telegram BOT
String BOTtoken = "";  // your Bot Token (Get from Botfather)

// Use @myidbot to find out the chat ID of an individual or a group
// Also note that you need to click "start" on a bot before it can
// message you
[... 188 lines cut ...]
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  clientTCP.setCACert(TELEGRAM_CERTIFICATE_ROOT); // Add root certificate for [api.telegram.org](http://api.telegram.org/)
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("ESP32-CAM IP Address: ");
  Serial.println(WiFi.localIP());
  bot.sendMessage(CHAT_ID, "Device ready!", "");
  
  sendPhotoTelegram(); 
  // if we're here, we wakon up
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_12,1); //1 = High, 0 = Low
  // let's try this deep sleep
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
}

void loop() {
}`

Thank you

FBMinis, 2024-01-23:
Regarding using deepsleep, if you want to save power, I suggest you keep the camera Off until there is movement detected by the PIR. By using one N-Mosfet (AO3400) and one P-Mosfet(AO3401),  the camera stays ON while the PIR output stays HIGH, during enough time to connect to the internet and send the photo to Telegram.

(everything is working at 3.3V in the project below)

[image]

[image]

[image]

[image]

FBMinis, 2024-01-23:
Replace:

```
camera_fb_t * fb = NULL;
fb = esp_camera_fb_get();
if (!fb) {
  Serial.println("Camera capture failed");
  delay(1000);
  ESP.restart();
  return "Camera capture failed";
}
```

With:

```
camera_fb_t * fb = NULL;
for (int j = 0; j < 7; j++) {
  camera_fb_t * newfb = esp_camera_fb_get();
  if (!newfb) {
    Serial.println("Camera Capture Failed");
  } else {
    esp_camera_fb_return(newfb);
    delay(10); //delay(10);
  }
}
fb = esp_camera_fb_get();
if (!fb) {
  Serial.println("Camera capture failed");
  bot.sendMessage(chat_id, "Camera capture failed", "");
  return;
}
```

FBMinis, 2024-01-23:
If you want to try a different sketch/library, this works for me. It sends a photo when powered ON.

```
#include <WiFi.h>

#include <AsyncTelegram2.h> //https://github.com/cotestatnt/AsyncTelegram2/tree/2.3.0

#include "esp_camera.h"
#include "soc/soc.h"           // Brownout error fix
#include "soc/rtc_cntl_reg.h"  // Brownout error fix

#include <WiFiClientSecure.h>
WiFiClientSecure secure_client;

const char* ssid = "your-ssid";                                        // SSID WiFi network
const char* pass = "your-password";                                        // Password  WiFi network
const char* token = "your-token";  // (Telegram token
// Check the userid with the help of bot @JsonDumpBot or @getidsbot (work also with groups)
// https://t.me/JsonDumpBot  or  https://t.me/getidsbot
int64_t userid = 0000000; //replace with your userid

int wifi_start_counter = 0;
// Timezone definition to get properly time from NTP server
#define MYTZ "WET0WEST,M3.5.0/1,M10.5.0" // Portugal
AsyncTelegram2 myBot(secure_client);

#define LED_PIN 33 // Pin 33 on ESP32CAM

#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27

#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
#define Y3_GPIO_NUM 18
#define Y2_GPIO_NUM 5
#define VSYNC_GPIO_NUM 25
#define HREF_GPIO_NUM 23
#define PCLK_GPIO_NUM 22

static camera_config_t camera_config = {
  .pin_pwdn = PWDN_GPIO_NUM,
  .pin_reset = RESET_GPIO_NUM,
  .pin_xclk = XCLK_GPIO_NUM,
  .pin_sscb_sda = SIOD_GPIO_NUM,
  .pin_sscb_scl = SIOC_GPIO_NUM,
  .pin_d7 = Y9_GPIO_NUM,
  .pin_d6 = Y8_GPIO_NUM,
  .pin_d5 = Y7_GPIO_NUM,
  .pin_d4 = Y6_GPIO_NUM,
  .pin_d3 = Y5_GPIO_NUM,
  .pin_d2 = Y4_GPIO_NUM,
  .pin_d1 = Y3_GPIO_NUM,
  .pin_d0 = Y2_GPIO_NUM,
  .pin_vsync = VSYNC_GPIO_NUM,
  .pin_href = HREF_GPIO_NUM,
  .pin_pclk = PCLK_GPIO_NUM,
  .xclk_freq_hz = 20000000,
  .ledc_timer = LEDC_TIMER_0,
  .ledc_channel = LEDC_CHANNEL_0,
  .pixel_format = PIXFORMAT_JPEG,  /* PIXFORMAT_RGB565, // for face detection/recognition */
  .frame_size = FRAMESIZE_VGA,
  .jpeg_quality = 12,
  .fb_count = 1,
[... 140 lines cut ...]
}

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);       // disable brownout detector

  Serial.begin(115200);
  Serial.println();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  blinkLED(2, 100); // blink red led  two times on power on

  init_wifi();

  sendToTelegram();

blinkLED(3, 100); // blink red led  three times when picture has been sent

}

void loop() {

}
```

feiticeir0, 2024-01-23:
Hi ! 
Thank you @FBMinis ! 
Will definitely try your method. Will keep you posted. 
But, still don't know why ESP32-Cam keeps turning the images white.

feiticeir0, 2024-01-25:
> Regarding using deepsleep, if you want to save power, I suggest you keep the camera Off until there is movement detected by the PIR. By using one N-Mosfet (AO3400) and one P-Mosfet(AO3401), the camera stays ON while the PIR output stays HIGH, during enough time to connect to the internet and send the photo to Telegram.
> 
> (everything is working at 3.3V in the project below)
> 
> ![IMG_20230309_125545](https://private-user-images.githubusercontent.com/36670323/299089671-cdb4a5e2-abe6-4368-8e31-f084c6a5860f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxODgzOTYsIm5iZiI6MTcwNjE4ODA5NiwicGF0aCI6Ii8zNjY3MDMyMy8yOTkwODk2NzEtY2RiNGE1ZTItYWJlNi00MzY4LThlMzEtZjA4NGM2YTU4NjBmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDEzMDgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0MDVjZDlhMTcyYTJhZWE0NGUxYTZjNDdiYzdmNWJlZmY5NDdiNDYzZTI1MjcwNzk5ZDgxMTEyNDE4YmVlMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fp5mKwPDRI17L69-tIiyZFkDX2WJklPjQ2k07NkGZbA)
> 
> ![IMG_20230309_125632](https://private-user-images.githubusercontent.com/36670323/299089747-ca44a4c6-391a-4e35-8ecc-8b1b32f33a2a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxODgzOTYsIm5iZiI6MTcwNjE4ODA5NiwicGF0aCI6Ii8zNjY3MDMyMy8yOTkwODk3NDctY2E0NGE0YzYtMzkxYS00ZTM1LThlY2MtOGIxYjMyZjMzYTJhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDEzMDgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMwOTA5Nzk1NjlmYjQxM2RlNzJlNjQ3MjJmMzczZmJiZDZmYjhlYTk2M2Y2NTU1ZjI1OGMzOGJlYTRiMmYwMWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OcqnO-Q_Tw4c8wBR3fKb3YzBSeFSSXhcUqTumjFx9i8)
> 
> ![IMG_20230114_155750](https://private-user-images.githubusercontent.com/36670323/299089533-93079013-e535-455e-a8b6-8f2d84a40314.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxODgzOTYsIm5iZiI6MTcwNjE4ODA5NiwicGF0aCI6Ii8zNjY3MDMyMy8yOTkwODk1MzMtOTMwNzkwMTMtZTUzNS00NTVlLWE4YjYtOGYyZDg0YTQwMzE0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDEzMDgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzOGJkZDczMWY5YjBiYWE1NzY2NDk3ZTdhYTc5MDE5NTc3MzRlZmZhNzBkZGIwMDA2MDlhYzU1NGE3N2I3ZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aHtMLsHhQj9sbKAIQkMMZxqQ8hgP9tYpWQ8d35DxwGI)
> 
> ![IMG_20230309_125437](https://private-user-images.githubusercontent.com/36670323/299089654-562025a7-4752-471e-b713-5493da7c27c4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxODgzOTYsIm5iZiI6MTcwNjE4ODA5NiwicGF0aCI6Ii8zNjY3MDMyMy8yOTkwODk2NTQtNTYyMDI1YTctNDc1Mi00NzFlLWI3MTMtNTQ5M2RhN2MyN2M0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDEzMDgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjNGY1MTJkMzlhNTg4NjljYmRhZjNjNjg5N2U5NjkxYzg2ZWI2ZjE4ZjE2MzY4Yjc4YjRkZGFmYzlmMGQ3YjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.p-eIsqG8PjHupy6l2f6enuC3wvN2BLJHqUdal8mU3dg)

Regarding your first post,  I'm using a PIR module connected to PIN 2 (if not mistaken) - and the ESP32 is in deep sleep. Only when theres a HIGH value on the PIN, it wakes up, initializes the camera, takes a picture and goes back to sleep. 

I would love to try your approach. can you send me a schematics ?

FBMinis, 2024-01-25:
[image]