espressif/esp32-camera #628: Can't capture a frame when PIXFORMAT_JPEG (Arduino IDE, CameraWebServer)

FBMinis, 2024-01-23:
Setup:

`Using library WiFi at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi`

[image]
[image]

When running the unmodified CameraWebServer sketch, I get the following message and no image is displayed:

`[ 10420][E][app_httpd.cpp:372] capture_handler(): Camera capture failed`

The OV2640 works with other boards.

When changing `PIXFORMAT_JPEG` to `PIXFORMAT_GRAYSCALE`, `PIXFORMAT_RGB565`, or `PIXFORMAT_YUV422`, I get the the following pictures, respectively.

[image]
[image]
[image]

Something is failing when `esp_camera_fb_get();` is called. **Is there a way to turn this function more verbose** so we can see what is happening in the SCCB interface? I don't have a logic analyzer...

I compared the values of some components between boards and I don't see any big discrepancies. 

I used this diagram to help me identify the components: https://easyeda.com/editor#id=5a6b4b9c6cde4d639ac2b8ab2624d384

Thank you

me-no-dev, 2024-01-24:
those boards are pretty much hit and miss. Many have different issues with them. If you say that the sensor works fine on other boards, but not this one, you should look for the issue in the board itself.

FBMinis, 2024-01-27:
Purchased a logic analyzer online, waiting for it to arrive. 

I noticed the sensors get hot when running on this board. Also, when running the camera webserver application  with other pixformat than jpeg, the board makes an audible noise coming from the esp32s chip.

It runs other examples from the arduino ide without a problem, for example the wifi stuff, sdcard and littlefs.

FBMinis, 2024-01-30:
This arrived after 4 business days, hope it will be helpful. It travelled 10000km for around €3.50 total.
[image]

closed 2024-02-07 by FBMinis