espressif/esp32-camera #630: ESP32S3 Xiao Sense crashes or fails to capture high quality picture

SYPark0795, 2024-01-28:
(I'm mostly copy-pasting a reddit post I made earlier)

First of all, I'm not interested in real time video, but in high quality still image.

When I try to capture a JPEG image with quality < 5 and (apparently) exceed the size of 350kb, the board mostly fails to capture the image (`esp_camera_fb_get()` returns 0) or otherwise the system crashes (shows debug messages and reboots). I have tested 3-4 boards, and they all exhibit the same problem.

When I google about esp32-S3 ov2640 high quality images, I see some mentioning about image corruption when the size exceeds some point, but there weren't any mention of failing to capture.

Here is the source code I ended up with (with a bit of help from ChatGPT): https://pastebin.com/9C1cLzXh - It doesn't have image transfer to anywhere because I'm trying to diagnose the capture issue with this code.

I have confirmed that PSRAM is enabled.

Another reddit user suggested me to print free heap size and free PSRAM, and both seem to have plenty of space remaining. They also told me to check the value of `CONFIG_CAMERA_CONVERTER_ENABLED`. VSCode could not find the definition of it, and it won't compile.

Additional question: Is it possible to capture 1600x1200 RGB565 or YUV422, even if it's slow?

SYPark0795, 2024-02-01:
I have ordered a few ESP32-CAM boards. I will compare how it behaves and update soon.

SYPark0795, 2024-02-03:
So far, I've been using Arduino's default version, and earlier today it occurred to me that, maybe using the most recent version of the library may fix the problem. I couldn't really confirm whether it actually does though. Could you tell me if there are possible fixes recently?

I only managed to compile esp32-camera library, and when I added the binary into the Arduino's library folder, it gave me the following error messages:  
```
E (6022) ledc: ledc_timer_config(556): speed_mode argument is invalid
[0;31mE (6023) camera_xclk: ledc_timer_config failed for freq 16000000, rc=102[0m
[0;31mE (6025) camera_xclk: ledc_timer_config failed, rc=102[0m
[0;31mE (6034) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)[0m
Camera initialization failed
```
Maybe I should properly set up esp-idf to make use of the most recent version of the library?

SYPark0795, 2024-02-04:
I ended up setting up ESP-IDF, and I think it's compiling the git version I put into the `components` folder... I'm still not capturing the HQ image, but I do see the debug messages  
```
Initializing the camera...
I (11902) s3 ll_cam: DMA Channel=0
I (11912) cam_hal: cam init ok
I (11912) sccb: pin_sda 40 pin_scl 39
I (11912) sccb: sccb_i2c_port=1
I (11932) camera: Detected camera at address=0x30
I (11932) camera: Detected OV2640 camera
I (11932) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (12022) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 375
I (12022) cam_hal: Allocating 384000 Byte frame buffer in PSRAM
I (12032) cam_hal: cam config ok
I (12032) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 12
W (12372) cam_hal: NO-EOI
W (12532) cam_hal: NO-EOI
W (12692) cam_hal: NO-EOI
W (12852) cam_hal: NO-EOI
W (13012) cam_hal: NO-EOI
W (13172) cam_hal: NO-EOI
W (13332) cam_hal: NO-EOI
W (13492) cam_hal: NO-EOI
W (13652) cam_hal: NO-EOI
W (13812) cam_hal: NO-EOI
W (13972) cam_hal: NO-EOI
W (14132) cam_hal: NO-EOI
W (14292) cam_hal: NO-EOI
W (14452) cam_hal: NO-EOI
W (14612) cam_hal: NO-EOI
W (14772) cam_hal: NO-EOI
W (14932) cam_hal: NO-EOI
W (15092) cam_hal: NO-EOI
W (15252) cam_hal: NO-EOI
W (15412) cam_hal: NO-EOI
W (15572) cam_hal: NO-EOI
W (15732) cam_hal: NO-EOI
W (15892) cam_hal: NO-EOI
W (16052) cam_hal: NO-EOI
W (16172) cam_hal: Failed to get the frame on time!
```
I realize it's allocating only 384000 bytes for the frame buffer. Maybe this is the problem, and if I could somehow allocate more for it, I could create bigger size photo out of it?

The following is part of the initialization.
```
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size = FRAMESIZE_UXGA;     // 1600x1200
  config.fb_count = 1;
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.jpeg_quality = 1;
```

SYPark0795, 2024-02-04:
I have played around with https://github.com/espressif/esp32-camera/blob/f0bb42917cddcfba2c32c2e2fb2875b4fea11b7a/driver/cam_hal.c#L376-L382 lines

In JPEG mode, I could allocate up to like 7MB to the frame buffer. However, it still fails to capture with the `NO-EOI` messages at a bit bigger than 300K.

In non-JPEG mode, it seems to have other problems on XGA (1024x768) and UXGA (1600x1200)
(Somehow it's fine on SXGA(1280 x 1024)) :
```
Heap: 370923/426579, 270336 max block
PSRAM: 8386308/8388608, 8257536 max block
I (6912) s3 ll_cam: DMA Channel=0
I (6912) cam_hal: cam init ok
I (6922) sccb: pin_sda 40 pin_scl 39
I (6922) sccb: sccb_i2c_port=1
I (6942) camera: Detected camera at address=0x30
I (6942) camera: Detected OV2640 camera
I (6942) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (7012) s3 ll_cam: node_size: 2048, nodes_per_line: 1, lines_per_node: 1
I (7012) s3 ll_cam: dma_half_buffer_min:  2048, dma_half_buffer: 16384, lines_per_half_buffer:  8, dma_buffer_size: 32768
I (7022) cam_hal: buffer_size: 32768, half_buffer_size: 16384, node_buffer_size: 2048, node_cnt: 16, total_cnt: 96
I (7032) cam_hal: Allocating 1572864 Byte frame buffer in PSRAM
I (7042) cam_hal: cam config ok
I (7042) ov2640: Set PLL: clk_2x: 1, clk_div: 7, pclk_auto: 1, pclk_div: 12
Heap: 334519/426579, 237568 max block
PSRAM: 6813228/8388608, 6684672 max block
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
E (11232) cam_hal: FB-SIZE: 933888 != 1572864
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
W (15112) cam_hal: Failed to get the frame on time!
```
Maybe it doesn't properly allocate the buffer?

me-no-dev, 2024-02-05:
`EV-VSYNC-OVF` would suggest that you are not getting proper data on the camera lines. Make sure that you have defined all data and control lines properly

SYPark0795, 2024-02-05:
1. There's more in the question than just `EV-VSYNC-OVF` part, that I think you might have missed. I have arbitrarily enlarged recv_size for JPEG mode, and **it's still not taking photos larger than some point a bit bigger than 300k in size, with the `NO-EOI` debug messages.** Under what circumstances does the library produce the `NO-EOI` messages?

2. As I mentioned in the very last post, **the same setting works perfectly fine for SXGA (1280x1024) but does not work for XGA (1024x768) and UXGA (1600x1200)** in RGB565 and YUV422 mode. Is it possible that I missed something and still success in capturing in SXGA, but fail in XGA?

Just in case, note that I have not transferred the pictures yet to confirm that the framebuffers contain valid pictures

SYPark0795, 2024-02-13:
Any update please? Any way I could help probing the problem, like turning on more debugging messages? Or maybe any source file I could look into? Any remarks? Should I just forget about esp32-s3 and use esp32-s instead, like ESP32-CAM? I've been checking for a comment every day since I left the last comment, and I even edited the comment for better wording, hoping I didn't offend you before :(

SYPark0795, 2024-03-01:
Just posted new issues for the two questions separately. Closing.

closed 2024-03-01 by SYPark0795