espressif/esp32-camera #633: BF3005 darker first picture

rbruneau-altyor, 2024-02-08:
Hi, 

I am working on a project that uses a BF3005 camera to take a single picture every time my ESP32 wakes up. The problem I am encountering is that it seems like the first picture I take is very dark, so I tried taking 3 pictures instead of a single one. It appears that the first picture avec the camera initialization is way darker than the following ones, as you can see on the following pictures:

1st picture after init: 
[image]

2nd picture after init:
[image]

How to correct this strange behavior ?

WangYuxin-esp, 2024-02-27:
Usually, the first few pictures after the device is powered on are used to correct the ISP parameters of the sensor. It is normal for these photos to have some problems. The simplest method is to actively call fb_return() to discard the first few frames of the image. Complex method: If it is possible to ensure that the surrounding environment of the sensor is consistent every time, information about the normal operation of the sensor in that scenario can be recorded, and this information can be written into the sensor every time it is powered on. We recommend that you proactively discard the first few frames of images to solve this problem, which usually does not require a lot of time.

rbruneau-altyor, 2024-02-27:
Noted, thanks for your answer, I will close it for now

closed 2024-02-27 by rbruneau-altyor