espressif/esp32-camera #635: BF3005 cam_hal: FB-OVF

rbruneau-altyor, 2024-02-09:
Hi,

I have a problem with my BF3005 camera application on ESP32-S3 (16MB flash, 8MB PSRAM). I use it the following way in application:

1. Initialize camera
2. Take a single picture
3. Deinitialize camera
4. Wait for a new picture and request and start over

Unfortunaltely, after 2 or 3 cycles, I get the following error, popping every 50ms:
`cam_hal: FB-OVF`

Here are my settings: 
```
.xclk_freq_hz = 10000000,
.ledc_timer = LEDC_TIMER_0,
.ledc_channel = LEDC_CHANNEL_0,
.pixel_format = PIXFORMAT_RGB565,
.frame_size = FRAMESIZE_QVGA, 
.jpeg_quality = 6,
.fb_count = 2
.grab_mode = CAMERA_GRAB_LATEST,
.fb_location = CAMERA_FB_IN_PSRAM
```

I have multiple other tasks that are running at the same time, but only 1 uses the BF3005 camera.
What is wrong here?

me-no-dev, 2024-02-10:
You do not need to restart the camera. Instead define `fb_count` to `1` and `grab_mode` to `CAMERA_GRAB_WHEN_EMPTY`. You return the old frame buffer and request a new one when a new frame is requested. That way the bus is in effect off when you do not need a frame.

rbruneau-altyor, 2024-02-12:
@me-no-dev actually I turn off the camera after each picture to save power, this is why I have to reinitialize it for each picture I take. Can I still use your parameters to avoid the `cam_hal: FB-OVF` error?

rbruneau-altyor, 2024-02-12:
Actually, even without restarting the camera, `fb_count = 1` and `grab_mode = CAMERA_GRAB_WHEN_EMPTY` still gives `cam_hal: FB-OVF` errors:

I (2156) s3 ll_cam: DMA Channel=4
I (2156) cam_hal: cam init ok
I (2156) sccb: pin_sda 4 pin_scl 5
I (2156) sccb: sccb_i2c_port=1
I (2166) camera: Detected camera at address=0x6e
I (2166) camera: Detected BF3005 camera
I (2166) camera: Camera PID=0x30 VER=0x03 MIDL=0x03 MIDH=0x30
I (2236) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (2236) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (2236) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (2246) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (2256) cam_hal: cam config ok
set_framesize 274
set_framesize 277
W (2426) cam_hal: FB-OVF
W (2526) cam_hal: FB-OVF

me-no-dev, 2024-02-12:
what are you changing the faremsize to?

rbruneau-altyor, 2024-02-12:
I only use `FRAMESIZE_QVGA` in my config before initialization

rbruneau-altyor, 2024-02-12:
Here is my code, that fits my hardware configuration: 

```
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// support IDF 5.x
#ifndef portTICK_RATE_MS
#define portTICK_RATE_MS portTICK_PERIOD_MS
#endif

#include "esp_camera.h"

#define SELF_HOLDING_PIN			GPIO_NUM_42
#define CAM_PIN_PWDN 				-1  		///< power down is not used
#define CAM_PIN_PWR	    			GPIO_NUM_21 ///<
#define CAM_PIN_RESET   			-1 			///< software reset will be performed
#define CAM_PIN_XCLK    			GPIO_NUM_15 ///<
#define CAM_PIN_SIOD    			GPIO_NUM_4  ///<
#define CAM_PIN_SIOC    			GPIO_NUM_5  ///<

#define CAM_PIN_D7      			GPIO_NUM_16 ///<
#define CAM_PIN_D6      			GPIO_NUM_17 ///<
#define CAM_PIN_D5      			GPIO_NUM_18 ///<
#define CAM_PIN_D4      			GPIO_NUM_12 ///<
#define CAM_PIN_D3      			GPIO_NUM_10 ///<
#define CAM_PIN_D2      			GPIO_NUM_8  ///<
#define CAM_PIN_D1      			GPIO_NUM_9  ///<
#define CAM_PIN_D0      			GPIO_NUM_11 ///<
#define CAM_PIN_VSYNC   			GPIO_NUM_6  ///<
#define CAM_PIN_HREF    			GPIO_NUM_7  ///<
#define CAM_PIN_PCLK    			GPIO_NUM_13 ///<

static const char *TAG = "example:take_picture";

#if ESP_CAMERA_SUPPORTED
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 10000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RGB565, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QVGA,    //QQVGA-UXGA, For ESP32, do not use sizes above QVGA when not JPEG. The performance of the ESP32-S series has improved a lot, but JPEG mode always gives better frame rates.

    .jpeg_quality = 6, //0-63, for OV series camera sensors, lower number means higher quality
    .fb_count = 1,       //When jpeg mode is used, if fb_count more than one, the driver will work in continuous mode.
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
	.fb_location = CAMERA_FB_IN_PSRAM
};

static esp_err_t init_camera(void)
{
    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }

    sensor_t *s = esp_camera_sensor_get();
    s->set_vflip(s, 1);//Flip the picture back
    s->set_hmirror(s, 1);//Mirror the picture

    return ESP_OK;
}
#endif

void app_main(void)
{
    uint8_t *image_data_buf = NULL;
    size_t image_data_buf_len = 0;
    esp_err_t _ret = 0;

#if ESP_CAMERA_SUPPORTED
    while (1)
    {
        if(ESP_OK != init_camera()) {
        	 ESP_LOGE(TAG, "Cloud not initialize camera");
            return;
        }
        esp_camera_fb_return(esp_camera_fb_get());

        vTaskDelay(1000 / portTICK_RATE_MS);

        ESP_LOGI(TAG, "Taking picture...");
        camera_fb_t *pic = esp_camera_fb_get();
        if (pic)
        {
			if (pic->format == PIXFORMAT_JPEG)
			{
				image_data_buf = pic->buf;
				image_data_buf_len = pic->len;
			}
			else if(!frame2jpg(pic, 80, &image_data_buf, &image_data_buf_len))
			{
				ESP_LOGE(TAG, "JPEG compression failed");
				_ret = ESP_FAIL;
			}
			if (_ret == ESP_OK)
			{
		        // use pic->buf to access the image
		        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
				for(uint32_t i = 0; i < image_data_buf_len; i++)
				{
					printf("%02x", image_data_buf[i]);
				}
				printf("\n");

				free(image_data_buf);
			}
        }
        esp_camera_fb_return(pic);
        esp_camera_deinit();
        vTaskDelay(5000 / portTICK_RATE_MS);
    }
#else
    ESP_LOGE(TAG, "Camera support is not available for this chip");
    return;
#endif
}
```

me-no-dev, 2024-02-12:
@WangYuxin-esp do you have BF3005 to test this?

WangYuxin-esp, 2024-02-18:
If there is only a slight `cam_hal: FB-OVF` error prompt and the image can be obtained normally after that, then simply ignore the error. This may be due to repeated power on and off causing sensor timing errors.

rbruneau-altyor, 2024-02-19:
Sometimes it's more like hundreds of `cam_hal: FB-OVF` error being logged. Until the camera is deinitialized. Also, I noticed that if I use  `xclk_freq_hz = 16MHz` instead of 10MHz, the error does not show up anymore.

me-no-dev, 2024-02-19:
> if I use  xclk_freq_hz = 16MHz instead of 10MHz, the error does not show up anymore.

this is because when set at 16MHz on S2 and S3, the EDMA mode is enabled and DMA writes directly to PSRAM. You will not get the OVF errors, because none of those are tracked anymore in EDMA mode, but that more is experimental and is not recommended as primary mode

rbruneau-altyor, 2024-02-19:
Ok, I will avoid using 16MHz then. What the errors could be due to in my application? It's quite a problem to get thousands of `cam_hal: FB-OVF` logs randomly. May be there is something wrong when I deinitialize/reinitialize the camera?

rbruneau-altyor, 2024-02-22:
@WangYuxin-esp any what I could improve?

WangYuxin-esp, 2024-02-27:
I have tested using BF3005. Tests have shown that when configured in xclk=20M, YUV422, QVGA or VGA mode, even if the device is repeatedly powered on and off, there are no abnormalities.  
In order to better identify the issue, I suggest that you check the VSYNC signal of the sensor through an oscilloscope or logic analyzer, and pay special attention to whether the signal is normal as usual when there is an abnormality.

rbruneau-altyor, 2024-02-27:
I will test with 20MHz and see what I get, thanks for your help.

According to a datasheet I found online, the right XCLK frequency should be 27MHz: 
[image]

So I tried this value during last week, and it seems to give way better results than I had with 10MHz. My question is, do 20MHz or 27MHz enable the experimental mode? As @me-no-dev recommended not to use it, may be I should avoid these values?

WangYuxin-esp, 2024-02-27:
As the input clock of the sensor, xclk affects many parameters. What I know is that the current BF3005 register configuration seems to be working at 20M xclk. In addition, only 16M will enable the ESP32S3 experimental mode.

rbruneau-altyor, 2024-02-27:
Ok, if I am not into experimental mode, I will stick to 20MHz or 27MHz with BF3005. Thank you

closed 2024-02-27 by rbruneau-altyor

WangYuxin-esp, 2024-02-27:
Note that xclk is obtained by dividing from a 160M clock source by default. Therefore, 27M clock cannot be accurately generated.

lulupalanche, 2026-01-16:
Moi, j'ai simplement constaté que l'erreur "cam_hal: FB-OVF" aparaissait lorsque la luminosité était faible
 I simply noticed that the error “cam_hal: FB-OVF” appeared when the brightness was low.

vitasam, 2026-10-28:
I just got the same `cam_hal: FB-OVF` error with esp32-s3 OV5640, when I close the camera with a dark cover. I have a theory that this could happen due to AEC (Auto Exposure Control) extends the exposure time per frame to capture more light. This drops the sensor's frame rate down to 5–10 FPS, causing the peripheral receives fluctuating frame lengths while the GDMA queue is active