espressif/esp32-camera #642: `cam_hal: NO-EOI` when trying to capture JPEG bigger than ~300KB

SYPark0795, 2024-03-01:
ESP-IDF version: v6.2, v6.1.3
Board: ESP32S3 Xiao Sense
Chip: ESP32-S3

Source code: https://pastebin.com/yyXjRYbj

When capture succeeds:
```
capturing in 10 seconds...
capturing in 9 seconds...
capturing in 8 seconds...
capturing in 7 seconds...
capturing in 6 seconds...
capturing in 5 seconds...
capturing in 4 seconds...
capturing in 3 seconds...
capturing in 2 seconds...
capturing in 1 seconds...
capturing in 0 seconds...
Camera Frame Information:
Width: 1600
Height: 1200
Format: 4
Size: 263476 bytes
Buffer: 0x3c050d70
```

When capture fails:
```
capturing in 10 seconds...
capturing in 9 seconds...
capturing in 8 seconds...
capturing in 7 seconds...
capturing in 6 seconds...
capturing in 5 seconds...
capturing in 4 seconds...
capturing in 3 seconds...
capturing in 2 seconds...
capturing in 1 seconds...
capturing in 0 seconds...
W (58658) cam_hal: NO-EOI
W (58878) cam_hal: NO-EOI
W (59038) cam_hal: NO-EOI
W (59198) cam_hal: NO-EOI
W (59358) cam_hal: NO-EOI
W (59518) cam_hal: NO-EOI
W (59668) cam_hal: NO-EOI
W (59838) cam_hal: NO-EOI
W (59998) cam_hal: NO-EOI
W (60158) cam_hal: NO-EOI
W (60318) cam_hal: NO-EOI
W (60478) cam_hal: NO-EOI
W (60638) cam_hal: NO-EOI
W (60798) cam_hal: NO-EOI
W (60948) cam_hal: NO-EOI
W (61108) cam_hal: NO-EOI
W (61268) cam_hal: NO-EOI
W (61428) cam_hal: NO-EOI
W (61588) cam_hal: NO-EOI
W (61748) cam_hal: NO-EOI
W (61908) cam_hal: NO-EOI
W (62068) cam_hal: NO-EOI
W (62228) cam_hal: NO-EOI
W (62388) cam_hal: NO-EOI
W (62548) cam_hal: NO-EOI
W (62648) cam_hal: Failed to get the frame on time!
Camera capture failed
```

I have played around with these lines to allocate more bytes to the framebuffer. It didn't help with the problem.
https://github.com/espressif/esp32-camera/blob/f0bb42917cddcfba2c32c2e2fb2875b4fea11b7a/driver/cam_hal.c#L376-L379

I have changed Default/Maximum Log Verbosity in menuconfig to Verbose. It only increased the amount of log during the boot, but nothing more comes out around the capturing part.

As far as I can tell, multiple modules I own exhibit the same behavior.

Please give me any insights you have. Where in the source code I could take a look, where I could probe more, etc. I could look into the library itself, too. I'd be happy to hear even just "I have no idea".

SYPark0795, 2024-04-30:
I'm still waiting for an answer.

SYPark0795, 2024-05-02:
What I want to achieve is to capture a picture with high quality, so setting the quality to a higher value is not the point for me.

When I set the quality to 3 or 2, it does capture a picture when I point the camera at something visually simple, so that the JPEG picture size goes lower. AFAICT, it's never the problem of the heat. It often captures a picture after several consecutive failed attempts, and it always fails to capture when the image goes complicated.

I have also tried ESP32-CAM and it's giving the same error.

me-no-dev, 2024-05-02:
If the sensor is not providing the proper bytes, there is no way for the ESP to verify the image. There is nothing wrong on our end, OV2640 and JPEG quality of 1 seems to not work correctly in the sensor itself. Have you tried getting RGB or YUYV image? Can't go higher quality than that

SYPark0795, 2024-05-02:
Yes I have. #643 is the problem I'm having with that.

closed 2024-07-10 by github-actions[bot]

pierluigizagaria, 2024-11-07:
@SYPark0795 have you found any solution? I would like to use the xiao board to capture high quality images, but seems to be impossible

xieyanname, 2025-01-17:
Hello, I also encountered a similar problem while taking photos with ESP32
pixel_format = PIXFORMAT_JPEG,
frame_size = FRAMESIZE_QSXGA,
jpeg_quality = 3,
The parameter I set is like this, it can take 925 KB photos, but adjusting the quality to 2 will prompt "NO-EOI"
May I ask if there is any solution?
The environment used is:
esp-idf v5.0.7
esp32-camera-v2.0.15

bakeromso, 2025-03-15:
I'm having the same issue. I can capture frame just fine until they reach over approximately 300kb. In menuconfig, I already set the custom frame size to 3M (I want to just capture one frame anyways), but getting NO-EOI on pictures that are too big