espressif/esp32-camera #646: ESP32S3-OV5640 can't update sensor settings

garvitkhera, 2024-03-08:
Hey @me-no-dev 

I hope you are doing well.

I am writing this post to ask for your help with an issue I'm having with OV5640 sensor settings.

I use a custom board with an ESP32-S3 chip connected to an OV5640 sensor. I'm using Platformio and trying to update the sensor settings like brightness, vFlip, and hMirror as usual:

```
// Initialize the camera
esp_err_t err = esp_camera_init(&config);
delay(1000);   // added a 1-second delay to let the camera sensor get stable (without this, the image quality was very unstable with a green tint
sensor_t *s = esp_camera_sensor_get();
s->set_hmirror(s, 1);
```
Even when using settings like this, nothing changes the image quality.
```
s->set_hmirror(s, 1);
//or 
s->set_brightness(s, 2);
```

I am trying to figure out what is causing this.

Below is my platformio.ini

```
[env:esp32cam]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

platform_packages = 
	platformio/tool-openocd-esp32

board_build.mcu = esp32s3
board_build.flash_mode = dio            
board_build.partitions = huge_app.csv   
board_upload.flash_size: 4MB
board_build.memory_type= dio_opi       
board_build.f_flash = 80000000L
board_build.f_cpu = 240000000L
board_upload.maximum_ram_size = 2097152     ; RAM - 2MB
board_upload.maximum_size = 4194304     ; FLASH - 4MB

build_unflags = -std=gnu++11

build_flags = 
        -std=gnu++17
        -DBOARD_HAS_PSRAM
        -DARDUINO_ESP32S3_DEV
        -DARDUINO_USB_CDC_ON_BOOT=1
        -DARDUINO_USB_MODE=1
lib_deps = 
	espressif/esp32-camera@^2.0.4
	bblanchon/ArduinoJson @ ~6.21.0
	fhessel/esp32_https_server @ ^1.0.0

upload_speed = 921600
monitor_speed = 115200
```

If you can suggest anything, I'd be grateful.

Thanks!

bulior, 2024-04-04:
push, have you solved it ?

garvitkhera, 2024-04-04:
Hi @bulior 
Adding a slight delay after `sensor_t *s = esp_camera_sensor_get();` solved the issue. I guess the sensor takes a little time to grab the current settings and then apply new ones.

garvitkhera, 2024-04-04:
Hi @bulior
Adding a slight delay after sensor_t *s = esp_camera_sensor_get(); solved the issue. I guess the sensor takes a little time to grab the current settings and then apply new ones.

closed 2024-04-04 by garvitkhera

bulior, 2024-04-04:
Thanks for your fast reply, but i will get the following error:
E (10027) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

Can you share your complete code ? Thanks ;)

hktalent, 2024-05-23:
The effect is very poor, why? Does anyone have a solution?
cat eye pattern behind door
what happened
[image]

OV5640
```
camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.frame_size = FRAMESIZE_SVGA; // UXGA|SXGA|XGA|SVGA|VGA|CIF|QVGA|HQVGA|QQVGA
  config.pixel_format = PIXFORMAT_JPEG;  // for streaming
  // config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 10;// 0-63 lower number means higher quality
  config.fb_count = 1;
  

  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
 if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){
      config.jpeg_quality = 10; // Lower value for higher quality
      config.fb_count = 3; // Increase frame buffer count
      config.grab_mode = CAMERA_GRAB_LATEST; // Change grab mode
    } else {
      // Increase the frame size when PSRAM is not available
      config.frame_size = FRAMESIZE_UXGA; // Higher resolution
      config.fb_location = CAMERA_FB_IN_DRAM;
    }
} else {
    // Best option for face detection/recognition
    config.frame_size = FRAMESIZE_240X240;
    #if CONFIG_IDF_TARGET_ESP32S3
        config.fb_count = 2;
    #endif
      }

    #if defined(CAMERA_MODEL_ESP_EYE)
      pinMode(13, INPUT_PULLUP);
      pinMode(14, INPUT_PULLUP);
    #endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  s->set_brightness(s, 1);     // Increase brightness
  s->set_contrast(s, 1);       // Increase contrast
  s->set_saturation(s, 1);     // Increase saturation
  s->set_special_effect(s, 0); // No special effect
  s->set_whitebal(s, 1);       // Enable white balance
  s->set_awb_gain(s, 1);       // Enable auto white balance gain
  s->set_wb_mode(s, 0);        // Set white balance mode to auto
  s->set_exposure_ctrl(s, 1);  // Enable exposure control
  s->set_aec2(s, 0);           // Disable automatic exposure control 2
  s->set_ae_level(s, 2);       // Increase auto exposure level
  s->set_aec_value(s, 803);    // Increase automatic exposure control value
  s->set_gain_ctrl(s, 1);      // Enable gain control
  s->set_agc_gain(s, 6);       // Increase automatic gain control level
  s->set_gainceiling(s, (gainceiling_t)6);  // Set gain ceiling to maximum
  s->set_bpc(s, 1);            // Enable black pixel cancellation
  s->set_wpc(s, 1);            // Enable white pixel cancellation
  s->set_raw_gma(s, 1);        // Enable raw gamma
  s->set_lenc(s, 1);           // Enable lens correction
  s->set_hmirror(s, 0);        // Disable horizontal mirror
  s->set_vflip(s, 0);          // Disable vertical flip
  s->set_dcw(s, 1);            // Enable downsize capture window
  s->set_colorbar(s, 0);       // Disable color bar

// If the sensor is OV5640, adjust the settings for this specific model
if (s->id.PID == OV5640_PID) {
  s->set_vflip(s, 1);        // flip it back
  s->set_brightness(s, 1);   // up the brightness just a bit
  s->set_saturation(s, -2);  // lower the saturation
}
// If the pixel format is JPEG, set the frame size to QVGA
// drop down frame size for higher initial frame rate
  if (config.pixel_format == PIXFORMAT_JPEG) {
    s->set_framesize(s, FRAMESIZE_SVGA);
  }

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

#if defined(CAMERA_MODEL_ESP32S3_EYE)
  s->set_vflip(s, 1);
#endif
```