espressif/esp32-camera #649: Support new I2C API in IDF 5.2

tannewt, 2024-03-21:
Hi! I'm working to update CircuitPython to IDF 5.2.1 and we use this library for camera support. We won't be able to use the new I2C API until this library is updated.

I can take a stab at it if you'd like. How would you like the two implementations separated?

larryli, 2024-06-07:
New I2C SCCB driver https://components.espressif.com/components/espressif/esp_sccb_intf

closed 2024-08-13 by github-actions[bot]

netroy, 2024-08-13:
stale-bot fails us all yet again. 
This should either be re-opened, or be closed as "not-planned" instead of closing as "completed".

playduck, 2024-08-25:
I've ported the SCCB driver code on my fork: https://github.com/playduck/esp32-camera (see PR #677 )
The camera driver now co-exists with other I2C devices using the newer API.
There's no breaking changes, this should be a drop-in replacement, however as of right now it needs the head version of the idf.

If you're using this as a managed idf compoennt you can simply change the definition in your project's `idf_component.yml` to point to my fork:
```yaml
dependencies:
  espressif/esp32-camera:
    git: https://github.com/playduck/esp32-camera.git
```

tannewt, 2024-08-28:
@igrr @me-no-dev Is this on your TODO list?

me-no-dev, 2024-08-29:
@tannewt yes.