espressif/esp32-camera #650: Use the example code, error in the following, how to solve, thank you

1308470235, 2024-04-02:
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:16576
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4284
entry 0x40080668
I (31) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (31) boot: compile time Apr  1 2024 18:02:49
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (82) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (89) boot:  2 factory          factory app      00 00 00020000 00100000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0ce48h ( 52808) map
I (124) esp_image: segment 1: paddr=0002ce70 vaddr=3ffb0000 size=03184h ( 12676) load
I (128) esp_image: segment 2: paddr=0002fffc vaddr=40080000 size=0001ch (    28) load
I (130) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=1e5d8h (124376) map
I (173) esp_image: segment 4: paddr=0004e600 vaddr=4008001c size=0e150h ( 57680) load
I (199) boot: Loaded app from partition at offset 0x20000
I (199) boot: Disabling RNG early entropy source...
I (211) cpu_start: Multicore app
I (211) cpu_start: Pro cpu up.
I (211) cpu_start: Starting app cpu, entry point is 0x40081290
0x40081290: call_start_cpu1 at D:/esp/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (229) cpu_start: Pro cpu start user code
I (229) cpu_start: cpu freq: 240000000 Hz
I (229) cpu_start: Application information:
I (234) cpu_start: Project name:     camera_example
I (239) cpu_start: App version:      1
I (244) cpu_start: Compile time:     Apr  1 2024 18:02:26
I (250) cpu_start: ELF file SHA256:  02da99ad58679f62...
I (256) cpu_start: ESP-IDF:          v5.1.2
I (261) cpu_start: Min chip rev:     v0.0
I (265) cpu_start: Max chip rev:     v3.99 
I (270) cpu_start: Chip rev:         v3.1
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (282) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (288) heap_init: At 3FFB3A30 len 0002C5D0 (177 KiB): DRAM
I (294) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (301) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (307) heap_init: At 4008E16C len 00011E94 (71 KiB): IRAM
I (314) spi_flash: detected chip: generic
I (318) spi_flash: flash io: qio
I (322) app_start: Starting scheduler on CPU0
I32327( 2pp_startt St:rting tehedn er 0 CPU1
I (332) main_task: Calling app_main()
I (336) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (345) cam_hal: cam init ok
I (349) sccb: pin_sda 26 pin_scl 27
I (353) sccb: sccb_i2c_port=1
I (356) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (395) camera: Detected camera at address=0x30
I (398) camera: Detected OV2640 camera
I (398) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (478) esp32 ll_cam: node_size: 2560, nodes_per_line: 1, lines_per_node: 1, dma_half_buffer_min:  2560, dma_half_buffer: 15360,lines_per_half_buffer:  6, dma_buffer_size: 30720, image_size: 153600
I (485) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 2560, node_cnt: 12, total_cnt: 10
I (496) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
E (502) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (509) cam_hal: cam_config(385): cam_dma_config failed
E (515) camera: Camera config failed with error 0xffffffff
E (521) example:take_picture: Camera Init Failed
I (526) main_task: Returned from app_main()

ahmedtariqh, 2024-04-06:
I have just the same error!
tried with arduino framework and the code works great!
it seems that the idf is missing something in the camera initialization

anticitizn, 2024-04-06:
Hello, I was just confronted with this as well. The cause is that the program is trying to allocate memory in PSRAM, but by default the PSRAM component is not included. You have to add `esp_psram` yourself like this:

./main/CMakeLists.txt
```
idf_component_register(SRCS take_picture.c
                        PRIV_INCLUDE_DIRS .
                        PRIV_REQUIRES nvs_flash esp_psram)
```

A minimal test to see if PSRAM is working:
```
#include <esp_psram.h>

size_t psram_size = esp_psram_get_size();
printf("PSRAM size: %d bytes\n", psram_size);
```

Also make sure to set your PSRAM to the correct mode (Octal or Quad, depending on your board) in `idf.py menuconfig` or the VSCode extension, or whatever you use to configure that. Espressif's development environment is a bit confusing..

anticitizn, 2024-04-06:
Another thing which seems to cause me grief is this line in the camera config in `take_picture.c`:

```
    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
```
My camera apparently didn't like the 20 MHz mode and only worked sporadically. When I changed it to `10000000` Hz, it started working fine.

1308470235, 2024-04-10:
The problem was solved by adding this parameter           .fb_location = CAMERA_FB_IN_DRAM,

anticitizn, 2024-04-10:
That solved your issue precisely because PSRAM was not enabled. DRAM however is quite small so you can only take quite small pictures, SVGA and below I believe.

1308470235, 2024-04-10:
> Hello, I was just confronted with this as well. The cause is that the program is trying to allocate memory in PSRAM, but by default the PSRAM component is not included. You have to add `esp_psram` yourself like this:你好，我刚刚也遇到了这个问题。原因是程序试图在 PSRAM 中分配内存，但默认情况下不包括 PSRAM 组件。你必须像这样添加自己：你好，我也刚刚遇到了这个问题。原因是程序试图在 PSRAM 中分配内存，但默认情况下不包括 PSRAM 组件。你必须这样添加自己：
> 
> ./main/CMakeLists.txt./main/CMakeLists.txt./main/CMakeLists.txt
> 
> ```
> idf_component_register(SRCS take_picture.c
>                         PRIV_INCLUDE_DIRS .
>                         PRIV_REQUIRES nvs_flash esp_psram)
> ```
> 
> A minimal test to see if PSRAM is working:检查 PSRAM 是否正常工作的最小测试：查看PSRAM是否正常工作的最小测试：
> 
> ```
> #include <esp_psram.h>
> 
> size_t psram_size = esp_psram_get_size();
> printf("PSRAM size: %d bytes\n", psram_size);
> ```
> 
> Also make sure to set your PSRAM to the correct mode (Octal or Quad, depending on your board) in `idf.py menuconfig` or the VSCode extension, or whatever you use to configure that. Espressif's development environment is a bit confusing..另请确保在 VSCode 扩展或任何用于配置的内容中将 PSRAM 设置为正确的模式（八进制或四进制，具体取决于您的主板）。乐鑫的开发环境有点混乱..此外，请确保将 PSRAM 设置为正确的模式（八进制或四进制，具体取决于您的主板）或 VSCode 扩展，或用于配置该扩展的任何内容。乐鑫的开发环境有点混乱。

thank you  Although I don’t understand much, I’m very grateful. The problem has been solved.

closed 2024-06-16 by github-actions[bot]

mrmonteith, 2024-10-11:
> The problem was solved by adding this parameter .fb_location = CAMERA_FB_IN_DRAM,

Sorry, but where did you add that to?

mpasteur, 2024-11-13:
> > That solved your issue precisely because PSRAM (slow but big external RAM) was not enabled. DRAM (small but fast internal ram) however is quite small so you can only take quite small pictures, SVGA and below I believe. If you need bigger resolutions try the fix above.这恰恰解决了您的问题，因为 PSRAM（缓慢但大的外部 RAM）未启用。然而，DRAM（小但快速的内部内存）非常小，因此您只能拍摄很小的照片，我相信是 SVGA 及以下。如果您需要更大的分辨率，请尝试上面的修复。
> 
> Can you write a simple example? I didn't understand how you solved it. I think you're right. I combined the ap example with the camera example and got an error again. I think it's the problem you mentioned.

That's no code. It's a configuration in you CMakeLists.txt file.
Something like:
```
idf_component_register(SRCS take_picture.c
                        PRIV_INCLUDE_DIRS .
                        PRIV_REQUIRES nvs_flash esp_http_server **esp_psram** esp_wifi)

```                        
 That's very important because if you don't do that and put image in DRAM you will suffer of lack of RAM soon if you want to make an http server or anything else

bnina-ayoub, 2025-01-21:
> The problem was solved by adding this parameter .fb_location = CAMERA_FB_IN_DRAM,

Hi @1308470235, Which file exactly you did add this field ?