espressif/esp32-camera #651: Increase Camera Capturing Time

neeloza-embrill, 2024-04-03:
Dear,

We are developing a Camera based application using ESP32-S3 (u-blox, NORA-W106-10B). The camera module OV5640 connected with ESP32-S3 using DVP (Parallel) interface.

It captures an image at "1280x720" (HD) resolution. Configuration of camera takes approximate 800-900 msec. We want to reduce this time as much as possible. As we check the source code after adding some debug print messages, we found that VSYNC interrupt is taking more than 600 msec after camera configuration and camera state at "CAM_STATE_IDLE".

Adding few logs to understand our concern.

```
I (523) camera: In esp_camera_init()
I (523) s3 ll_cam: DMA Channel=4
I (523) camera: Before camera_probe()
I (523) sccb: pin_sda 10 pin_scl 5
I (533) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (533) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (533) camera: Searching for camera address...
I (543) ov3660: Mismatch PID=0x5640
I (543) camera: Detected OV5640 camera
I (543) camera: Camera PID=0x5640 VER=0x00 MIDL=0x00 MIDH=0x00
I (553) camera: Before s_state->sensor.reset()
I (793) camera: After s_state->sensor.reset()
I (793) camera: After camera_probe()
I (793) camera: Before cam_config()
I (793) cam_hal: cam_config
I (793) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3
I (793) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 12, dma_buffer_size: 30720
I (793) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 60
I (793) cam_hal: cam config ok
I (793) camera: After cam_config()
I (813) ov5640: Set PLL: bypass: 0, multiplier: 10, sys_div: 1, pre_div: 2, root_2x: 0, pclk_root_div: 1, pclk_manual: 1, pclk_div: 2
I (813) ov5640: Calculated XVCLK: 20000000 Hz, REFIN: 10000000 Hz, VCO: 100000000 Hz, PLL_CLK: 40000000 Hz, SYSCLK: 10000000 Hz, PCLK: 10000000 Hz
I (853) APP_MAIN: Camera Configuration Success
I (873) cam_hal: Before xQueueReceive(cam_obj->frame_buffer_queue)
ll_cam_vsync_isr INTERUUPT
I (1203) cam_hal: cam_task CAM_STATE_IDLE
I (1203) cam_hal: cam_event == CAM_VSYNC_EVENT ************************************************************
ll_cam_vsync_isr INTERUUPT
I (1403) cam_hal: cam_task CAM_STATE_READ_BUF
I (1403) cam_hal: After xQueueReceive(cam_obj->frame_buffer_queue)
I (1403) APP_MAIN: ++capturing_count-> 0
I (1413) cam_hal: Before xQueueReceive(cam_obj->frame_buffer_queue)
ll_cam_vsync_isr INTERUUPT
I (1603) cam_hal: cam_task CAM_STATE_READ_BUF
I (1603) cam_hal: After xQueueReceive(cam_obj->frame_buffer_queue)
I (1603) APP_MAIN: ++capturing_count-> 1
I (1613) cam_hal: Before xQueueReceive(cam_obj->frame_buffer_queue)
ll_cam_vsync_isr INTERUUPT
I (1793) cam_hal: cam_task CAM_STATE_READ_BUF
I (1793) cam_hal: After xQueueReceive(cam_obj->frame_buffer_queue)
I (1793) APP_MAIN: ++capturing_count-> 2
```

Looking forward your response. Thanks in advance.

me-no-dev, 2024-04-04:
The sensors themselves have MCU with ISP inside. I would guess that it takes a while for everything to get reconfigured inside and to issue VSYNC with a new frame

neeloza-embrill, 2024-04-05:
Hello @me-no-dev ,

Thanks for your quick response. Will you please suggest camera module or sensor to increase capturing time?

me-no-dev, 2024-04-05:
@neeloza-embrill I have actually never tested how much time is necessary for sensor startup, so I can not comment on that. If you need JPEG, then you are limited to OV2640, OV3360 or OV5640

closed 2024-06-08 by neeloza-embrill