espressif/esp32-camera #655: cam_hal:   insufficient queue_size in corner case.

Hello1024, 2024-04-24:
The `cam_obj->event_queue` is initialized based on a queue_size calculated from the `dma_half_buffer_cnt - 1`.

However, as well as processing DMA buffers, this queue also processes the vsync events.    If, due to bad luck, the vsync event is fired at the exact same time as a buffer is filled, then the queue overflows.   This leads to frame capture failing every time.   The camera can produce a VSYNC event just one clock cycle after sending the last data byte, so this happens rather frequently.

The fix is to remove the "-1" from the formula for queue_size.

One reproducible case is the OV2640 camera running with an XCLK of 20 Mhz, `PIXFORMAT_GRAYSCALE`, and `FRAMESIZE_QVGA`.  (all other options the same as the `camera_example`)

AxelLin, 2024-05-09:
> One reproducible case is the OV2640 camera running with an XCLK of 20 Mhz, `PIXFORMAT_GRAYSCALE`, and `FRAMESIZE_QVGA` on an ESP32C3 with PSRAM. (all other options the same as the `camera_example`)

@Hello1024 
Just wondering if you really have a **ESP32C3 with PSRAM** ?  Is C3 a typo?

Hello1024, 2024-07-26:
For anyone else looking, the "+1" does fix the bug, but introduces others I think, because it is then possible for a data buffer to get reused if there *isn't* a VSYNC, but the cam_task doesn't get run.  (this would never produce valid data, but the frame isn't correctly dropped as it should be).

A different fix is required - and I don't have plans to work on it, because this patch as-is works for my use case.