espressif/esp32-camera #657: Enable psram mode on ESP32S3 causes NO-SOI and then panic

AxelLin, 2024-05-09:
Test esp32-cam-v2.0.8 with esp-idf v5.2.1-485-g9b3ae86417.
Enable "psram mode" by setting xclk_freq_hz = 16000000.
Then I got below panic.

Note: it works if setting xclk_freq_hz = 20000000.

```
I (6278) s3 ll_cam: DMA Channel=1
I (6278) cam_hal: cam init ok
I (6279) sccb: pin_sda 18 pin_scl 8
I (6279) sccb: sccb_i2c_port=1
I (6293) camera: Detected camera at address=0x30
I (6296) camera: Detected OV2640 camera
I (6296) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (6380) cam_hal: buffer_size: 384000, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 375, total_cnt: 375
I (6381) cam_hal: Allocating 384016 Byte frame buffer in PSRAM
I (6387) cam_hal: Frame[0]: Offset: 16, Addr: 0x3C120500
I (6393) cam_hal: cam config ok
I (6397) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 12
W (6634) cam_hal: NO-SOI
W (6734) cam_hal: NO-SOI
W (6834) cam_hal: NO-SOI
W (6933) cam_hal: NO-SOI
W (7033) cam_hal: NO-SOI
W (7133) cam_hal: NO-SOI
W (7233) cam_hal: NO-SOI
W (7333) cam_hal: NO-SOI
W (7433) cam_hal: NO-SOI
W (7533) cam_hal: NO-SOI
W (7633) cam_hal: NO-SOI
W (7733) cam_hal: NO-SOI
W (7833) cam_hal: NO-SOI
W (7933) cam_hal: NO-SOI
W (8033) cam_hal: NO-SOI
W (8133) cam_hal: NO-SOI
W (8233) cam_hal: NO-SOI
W (8333) cam_hal: NO-SOI
W (8433) cam_hal: NO-SOI
W (8533) cam_hal: NO-SOI
W (8633) cam_hal: NO-SOI
W (8732) cam_hal: NO-SOI
W (8832) cam_hal: NO-SOI
W (8932) cam_hal: NO-SOI
W (9032) cam_hal: NO-SOI
W (9132) cam_hal: NO-SOI
W (9232) cam_hal: NO-SOI
W (9332) cam_hal: NO-SOI
W (9432) cam_hal: NO-SOI
W (9532) cam_hal: NO-SOI
W (9632) cam_hal: NO-SOI
W (9732) cam_hal: NO-SOI
W (9832) cam_hal: NO-SOI
W (9932) cam_hal: NO-SOI
W (10032) cam_hal: NO-SOI
W (10132) cam_hal: NO-SOI
W (10232) cam_hal: NO-SOI
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40376695  PS      : 0x00060033  A0      : 0x403777d8  A1      : 0x3fc9c450  
0x40376695: shared_intr_isr at /home/axel/esp/esp-idf/components/esp_hw_support/intr_alloc.c:443 (discriminator 1)
0x403777d8: _xt_medint2 at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:1329

A2      : 0x3fcca340  A3      : 0x0000cdcd  A4      : 0x3fc9f314  A5      : 0x4037e986  
0x4037e986: _frxt_int_enter at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:140

A6      : 0x00000000  A7      : 0x3fccc1f8  A8      : 0x8037e865  A9      : 0x3fc9c420  
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x00000000  A13     : 0x00060023  
A14     : 0x01ffffff  A15     : 0x0000cdcd  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8037e865  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  

Backtrace: 0x40376692:0x3fc9c450 0x403777d5:0x3fc9c470 0x400559dd:0x3fccc2b0 0x4037e862:0x3fccc2c0 0x4037e3f9:0x3fccc2e0 0x4037e4b8:0x3fccc320 0x40376b3d:0x3fccc340 0x40376c05:0x3fccc370 0x420c388e:0x3fccc390 0x42008516:0x3fccc3c0 0x420c2c91:0x3fccc3e0 0x420a52ae:0x3fccc400 0x420a4501:0x3fccc420 0x420a455e:0x3fccc440 0x420a4db9:0x3fccc470 0x420b6137:0x3fccc4a0 0x420b5afa:0x3fccc4c0 0x420a5325:0x3fccc7e0 0x420c25fd:0x3fccc810 0x40384846:0x3fccc840 0x42016dd5:0x3fccc890
0x40376692: shared_intr_isr at /home/axel/esp/esp-idf/components/esp_hw_support/intr_alloc.c:443 (discriminator 1)
0x403777d5: _xt_medint2 at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:1329
0x4037e862: vPortClearInterruptMaskFromISR at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:564
 (inlined by) vPortExitCritical at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:504
0x4037e3f9: xQueueSemaphoreTake at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1777
0x4037e4b8: xQueueTakeMutexRecursive at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:835
0x40376b3d: lock_acquire_generic at /home/axel/esp/esp-idf/components/newlib/locks.c:144
0x40376c05: _lock_acquire_recursive at /home/axel/esp/esp-idf/components/newlib/locks.c:158
0x420c388e: uart_write at /home/axel/esp/esp-idf/components/vfs/vfs_uart.c:209
0x42008516: console_write at /home/axel/esp/esp-idf/components/vfs/vfs_console.c:73 (discriminator 1)
0x420c2c91: esp_vfs_write at /home/axel/esp/esp-idf/components/vfs/vfs.c:482 (discriminator 4)
0x420a52ae: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x420a4501: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x420a455e: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
 (inlined by) _fflush_r at /builds/idf/crosstool-NG/.build/src/newlib-git-af24ceb7/newlib/libc/include/stdio.h:417
0x420a4db9: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x420b6137: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x420b5afa: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1774 (discriminator 1)
0x420a5325: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x420c25fd: esp_log_writev at /home/axel/esp/esp-idf/components/log/log.c:212
0x40384846: esp_log_write at /home/axel/esp/esp-idf/components/log/log.c:222
0x42016dd5: cam_verify_jpeg_soi at /home/axel/esp/esp-idf-dev/apps/ins-cam/managed_components/espressif__esp32-camera/driver/cam_hal.c:56 (discriminator 1)
 (inlined by) cam_task at /home/axel/esp/esp-idf-dev/apps/ins-cam/managed_components/espressif__esp32-camera/driver/cam_hal.c:159 (discriminator 1)

```

AxelLin, 2024-07-09:
I tested again with esp32-cam-v2.0.8 and esp-idf v5.2.2-301-g0250fc5cd4.
I no longer hit the panic, but I got a lot of "W (43004) cam_hal: NO-EOI".
I'm testing with ov2640.

mmalecki, 2024-07-11:
Hm, oddly enough, I'm hitting this one right now, but on only one of two same boards. The other one works perfectly:

```
I (1618) s3 ll_cam: DMA Channel=1
I (1618) cam_hal: cam init ok
I (1618) sccb: pin_sda 4 pin_scl 5
I (1628) sccb: sccb_i2c_port=1
I (1628) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1658) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1688) camera: Detected camera at address=0x30
I (1688) camera: Detected OV2640 camera
I (1688) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1768) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 60
I (1768) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1778) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1788) cam_hal: cam config ok
I (1788) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
W Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x403767d8  PS      : 0x00060033  A0      : 0x40377d48  A1      : 0x3fc9cd00  
0x403767d8: shared_intr_isr at /home/maciej/dev/src/esp-idf/components/esp_hw_support/intr_alloc.c:440
0x40377d48: _xt_lowint1 at /home/maciej/dev/src/esp-idf/components/xtensa/xtensa_vectors.S:1240

A2      : 0x3fcd5c28  A3      : 0x3fcac8e4  A4      : 0x20000000  A5      : 0x4037f322  
0x4037f322: _frxt_int_enter at /home/maciej/dev/src/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:140

A6      : 0x60023000  A7      : 0x0000000d  A8      : 0x00000000  A9      : 0x60023000  
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x8037ed27  A13     : 0x00060023  
A14     : 0x3fcac930  A15     : 0x0000cdcd  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000d  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  
0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x403767d5:0x3fc9cd00 0x40377d45:0x3fc9cd20 0x40383e19:0x3fcdc030 0x403773fe:0x3fcdc050 0x4200a0bb:0x3fcdc070 0x4200a2d6:0x3fcdc0a0 0x42009e09:0x3fcdc0c0 0x420094ce:0x3fcdc0e0 0x420a1e66:0x3fcdc100 0x420a1211:0x3fcdc120 0x420a126e:0x3fcdc140 0x420a18ad:0x3fcdc170 0x420b0c57:0x3fcdc1a0 0x420b061a:0x3fcdc1c0 0x420a1edd:0x3fcdc4e0 0x420bd841:0x3fcdc510 0x4200b786:0x3fcdc540 0x4200b809:0x3fcdc580 0x4201b01b:0x3fcdc5d0 0x4201b8ed:0x3fcdc5f0 0x4037eef2:0x3fcdc630
0x403767d5: vPortEnterCritical at /home/maciej/dev/src/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:572
 (inlined by) shared_intr_isr at /home/maciej/dev/src/esp-idf/components/esp_hw_support/intr_alloc.c:438
0x40377d45: _xt_lowint1 at /home/maciej/dev/src/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x40383e19: systimer_ll_is_counter_value_valid at /home/maciej/dev/src/esp-idf/components/hal/esp32s3/include/hal/systimer_ll.h:65
 (inlined by) systimer_hal_get_counter_value at /home/maciej/dev/src/esp-idf/components/hal/systimer_hal.c:51
0x403773fe: esp_timer_impl_get_time at /home/maciej/dev/src/esp-idf/components/esp_timer/src/esp_timer_impl_systimer.c:72
0x4200a0bb: usb_serial_jtag_tx_char at /home/maciej/dev/src/esp-idf/components/vfs/vfs_usb_serial_jtag.c:123
0x4200a2d6: usb_serial_jtag_write at /home/maciej/dev/src/esp-idf/components/vfs/vfs_usb_serial_jtag.c:156
0x42009e09: console_write at /home/maciej/dev/src/esp-idf/components/vfs/vfs_console.c:75 (discriminator 1)
0x420094ce: esp_vfs_write at /home/maciej/dev/src/esp-idf/components/vfs/vfs.c:482 (discriminator 4)
0x420a1e66: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x420a1211: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x420a126e: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
 (inlined by) _fflush_r at /builds/idf/crosstool-NG/.build/src/newlib-git-af24ceb7/newlib/libc/include/stdio.h:417
0x420a18ad: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x420b0c57: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfiprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x420b061a: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1774 (discriminator 1)
0x420a1edd: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x420bd841: esp_log_writev at /home/maciej/dev/src/esp-idf/components/log/log.c:215
0x4200b786: __wrap_esp_log_writev at /home/maciej/dev/drones/DroneBridge-ESP32/managed_components/espressif__esp-now/src/debug/src/espnow_log.c:110
0x4200b809: __wrap_esp_log_write at /home/maciej/dev/drones/DroneBridge-ESP32/managed_components/espressif__esp-now/src/debug/src/espnow_log.c:138
0x4201b01b: cam_verify_jpeg_soi at /home/maciej/dev/drones/DroneBridge-ESP32/managed_components/espressif__esp32-camera/driver/cam_hal.c:56 (discriminator 1)
0x4201b8ed: cam_task at /home/maciej/dev/drones/DroneBridge-ESP32/managed_components/espressif__esp32-camera/driver/cam_hal.c:159 (discriminator 2)
0x4037eef2: vPortTaskWrapper at /home/maciej/dev/src/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

closed 2024-09-18 by github-actions[bot]

AgainPsychoX, 2024-12-21:
I have the same issue I think. 

Here is longer log, since there are few different variants of the crash: [device-monitor-241221-105104.log](https://github.com/user-attachments/files/18218096/device-monitor-241221-105104.log)