espressif/esp32-camera #660: The example code cannot run properly (LoadProhibited)

MikeBrownse, 2024-05-29:
ESP-IDF version: v5.2.1 (release version)
Chip: ESP32-S3-WROOM-1-N16R8

When running the example code "take_picture.c", the following error shows, and reboot again and again.
So, is there any wrong with the file "ll_cam.c"?

`
......
I (31) boot: ESP-IDF v5.2 2nd stage bootloader
I (31) boot: compile time May 29 2024 21:56:39
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32s3: Boot SPI Speed : 80MHz
I (48) boot.esp32s3: SPI Mode       : QIO
I (53) boot.esp32s3: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (81) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (89) boot:  2 factory          factory app      00 00 00020000 00100000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00020020 vaddr=3c030020 size=0f8e0h ( 63712) map
I (119) esp_image: segment 1: paddr=0002f908 vaddr=3fc94d00 size=00710h (  1808) load
I (119) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=241ech (147948) map
I (148) esp_image: segment 3: paddr=00054214 vaddr=3fc95410 size=035bch ( 13756) load
I (151) esp_image: segment 4: paddr=000577d8 vaddr=40374000 size=10bf8h ( 68600) load
I (174) boot: Loaded app from partition at offset 0x20000
I (174) boot: Disabling RNG early entropy source...
I (186) cpu_start: Multicore app
I (186) octal_psram: vendor id    : 0x0d (AP)
I (186) octal_psram: dev id       : 0x02 (generation 3)
I (189) octal_psram: density      : 0x03 (64 Mbit)
I (195) octal_psram: good-die     : 0x01 (Pass)
I (200) octal_psram: Latency      : 0x01 (Fixed)
I (205) octal_psram: VCC          : 0x01 (3V)
I (210) octal_psram: SRF          : 0x01 (Fast Refresh)
I (216) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (222) octal_psram: BurstLen     : 0x01 (32 Byte)
I (227) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (234) octal_psram: DriveStrength: 0x00 (1/1)
I (239) MSPI Timing: PSRAM timing tuning index: 5
I (244) esp_psram: Found 8MB PSRAM device
I (249) esp_psram: Speed: 80MHz
I (705) esp_psram: SPI SRAM memory test OK
I (714) cpu_start: Pro cpu start user code
I (714) cpu_start: cpu freq: 240000000 Hz
I (714) cpu_start: Application information:
I (717) cpu_start: Project name:     camera_example
I (722) cpu_start: App version:      1
I (727) cpu_start: Compile time:     May 29 2024 21:57:32
I (733) cpu_start: ELF file SHA256:  3c690e871...
I (738) cpu_start: ESP-IDF:          v5.2
I (743) cpu_start: Min chip rev:     v0.0
I (748) cpu_start: Max chip rev:     v0.99
I (752) cpu_start: Chip rev:         v0.1
I (757) heap_init: Initializing. RAM available for dynamic allocation:
I (764) heap_init: At 3FC992E0 len 00050430 (321 KiB): RAM
I (771) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (777) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (783) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (789) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (797) spi_flash: detected chip: gd
I (801) spi_flash: flash io: qio
W (805) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (815) sleep: Configure to isolate all GPIO pins in sleep state
I (822) sleep: Enable automatic switching of GPIO sleep configuration
I (830) main_task: Started on CPU0
I (834) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (843) main_task: Calling app_main()
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4201092a  PS      : 0x00060230  A0      : 0x82007c1a  A1      : 0x3fc9c8e0  
0x4201092a: ll_cam_set_pin at C:/workSpaces/VSProjects/esp32-camera/camera_example/managed_components/espressif__esp32-camera/target/esp32s3/ll_cam.c:366

A2      : 0x3c03b96c  A3      : 0x3fc94e60  A4      : 0x00000008  A5      : 0x00000000
A6      : 0x3fc9c810  A7      : 0x3fca5d00  A8      : 0x3c03b9c4  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000060  A12     : 0x00000008  A13     : 0x00000000
A14     : 0x3c0331a2  A15     : 0x00000060  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x42010927:0x3fc9c8e0 0x42007c17:0x3fc9c920 0x42007523:0x3fc9c950 0x42007256:0x3fc9c990 0x4200727b:0x3fc9c9b0 0x42023303:0x3fc9c9d0 0x4037c991:0x3fc9ca00
0x42010927: ll_cam_set_pin at C:/workSpaces/VSProjects/esp32-camera/camera_example/managed_components/espressif__esp32-camera/target/esp32s3/ll_cam.c:366
0x42007c17: cam_init at C:/workSpaces/VSProjects/esp32-camera/camera_example/managed_components/espressif__esp32-camera/driver/cam_hal.c:340
0x42007523: esp_camera_init at C:/workSpaces/VSProjects/esp32-camera/camera_example/managed_components/espressif__esp32-camera/driver/esp_camera.c:275
0x42007256: init_camera at C:/workSpaces/VSProjects/esp32-camera/camera_example/main/take_picture.c:154
0x4200727b: app_main at C:/workSpaces/VSProjects/esp32-camera/camera_example/main/take_picture.c:168
0x42023303: main_task at C:/Users/25347/esp/v5.2/esp-idf/components/freertos/app_startup.c:208
0x4037c991: vPortTaskWrapper at C:/Users/25347/esp/v5.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 3c690e871

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b34
0x40375b34: esp_restart_noos at C:/Users/25347/esp/v5.2/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x3024
entry 0x403c993c
I (31) boot: ESP-IDF v5.2 2nd stage bootloader
I (31) boot: compile time May 29 2024 21:56:39
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32s3: Boot SPI Speed : 80MHz
I (48) boot.esp32s3: SPI Mode       : QIO
I (53) boot.esp32s3: SPI Flash Size : 16MB
......
`

MikeBrownse, 2024-05-29:
The camera is ov5640, and it seems going wrong when setting the GPIO PINs in "ll_cam.c", line 366.
NEED HELP!!!😭

me-no-dev, 2024-05-30:
What is your camera configuration?

MikeBrownse, 2024-05-30:
> What is your camera configuration?

the model is "[alientek OV5640 module](http://www.openedv.com/docs/modules/camera/ov5640.html)", with a 24MHz oscillator connected to XCLK
picture of the module:
[image]

schematic diagram of the module:
[image]

MikeBrownse, 2024-05-30:
> and how about your camera driver configuration? In the code. Where you define the pins that the camera uses

the pins are defined as follows:
```
#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 21
#define CAM_PIN_SIOD 28 // 26
#define CAM_PIN_SIOC 27

#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 19
#define CAM_PIN_D2 18
#define CAM_PIN_D1 5
#define CAM_PIN_D0 4
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22
```

me-no-dev, 2024-05-30:
You can not use GPIOs 34 and 35 on ESP32-S3 with Octal PSRAM. Change the pins to something else

me-no-dev, 2024-05-30:
make sure that you do not use 36 and 37 also.

closed 2024-05-30 by me-no-dev

MikeBrownse, 2024-05-30:
I've changed the pins, but still not work:

```
#define CAM_PIN_PWDN  -1 //power down is not used
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK  17

#define CAM_PIN_SIOD 5
#define CAM_PIN_SIOC 4

#define CAM_PIN_D7 25
#define CAM_PIN_D6 24
#define CAM_PIN_D5 23
#define CAM_PIN_D4 22
#define CAM_PIN_D3 21
#define CAM_PIN_D2 20
#define CAM_PIN_D1 19
#define CAM_PIN_D0 18

#define CAM_PIN_VSYNC 12
#define CAM_PIN_HREF 13
#define CAM_PIN_PCLK 14
```

MikeBrownse, 2024-05-30:
> make sure that you do not use 36 and 37 also.

and could u tell me where I can find out the special pins of the chip used?

me-no-dev, 2024-05-30:
Sorry, but where do you see GPIOs 22-25 on the module? Make sure you match the GPIO numbers not the physical pin number on the module

me-no-dev, 2024-05-30:
> and could u tell me where I can find out the special pins of the chip used?

In the module datasheet. https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf

MikeBrownse, 2024-05-30:
> Sorry, but where do you see GPIOs 22-25 on the module? Make sure you match the GPIO numbers not the physical pin number on the module

oh! I see! I think I did fill in the physical pin numbers mistakenly.
Thank you for your patience!

icemanexi, 2024-07-02:
not relavent to original problem, but I ran into this thread while fixing mine. Using a esp32 s3 wroom-1 freenove kit, with ov2640. pinout was:

#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 15
#define CAM_PIN_SIOD 4
#define CAM_PIN_SIOC 5

// do not use pins 34 35 36 36

#define CAM_PIN_D7 11
#define CAM_PIN_D6 9
#define CAM_PIN_D5 8
#define CAM_PIN_D4 10
#define CAM_PIN_D3 12
#define CAM_PIN_D2 18
#define CAM_PIN_D1 17
#define CAM_PIN_D0 16
#define CAM_PIN_VSYNC 6
#define CAM_PIN_HREF 7
#define CAM_PIN_PCLK 13

I sourced this from the freenove git package: https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board/
under main/C/Sketches/Sketch_07.1_CameraWebServer/camera_pins.h, using CAMERA_MODEL_ESP32S3_EYE.

some additional settings I did just so everything is on one post: 
set psram to octal
spi ram speed 80 MHz