espressif/esp32-camera #663: Hardcoded buffer size for JPEG is too small for low resolution settings (e.g., QQVGA)

maruta, 2024-06-16:
### Description:

When using JPEG mode with low resolution settings such as QQVGA, the allocated frame buffer size frequently becomes insufficient, causing capture failures. This issue arises due to the hardcoded buffer size calculation in cam_hal.h.

https://github.com/espressif/esp32-camera/blob/30aeeeed61e2b0182d673fe2cfedb1b7aa43b1b9/driver/cam_hal.c#L376-L379

I have confirmed this problem with OV2640. The settings at that time were as follows
``` c
config.pixel_format = PIXFORMAT_JPEG;
config.frame_size = FRAMESIZE_QQVGA;
config.jpeg_quality = 16;
```
The hardcoded limit for this setting is 3840 bytes, but this limit is frequently exceeded when capturing complex objects, even with a relatively low JPEG quality setting.

### Suggested Fix:

One potential solution is to add a new member to the `camera_config_t` structure to allow the frame buffer size to be adjusted during the `esp_camera_init` call. However, I am not very experienced with this library, so this suggestion might not be the most feasible.

me-no-dev, 2024-06-17:
There is no clear way to calculate the expected maximum size that an image can be. Spec goes as far to say that with high quality, the image can be larger than it's RGB counterpart. I'll think about providing a way to adjust this, or make it slightly larger for smaller resolutions

maruta, 2024-07-17:
This issue has been resolved by #667

closed 2024-07-17 by maruta