espressif/esp32-camera #664: Support for Higher Resolutions on gc2145 - $300 Reward

MaxMaeder, 2024-06-28:
Hello,

As previously raised in the issue #122, the current driver for the GC2145 sensor does not support high resolutions, but it was stated that with modifications to the default register configuration for the sensor, this should be possible.

I need to capture images from the sensor at FRAMESIZE_UXGA and FRAMESIZE_HD resolutions. This is a critical business concern, so I am offering a US $300 reward to anyone who can make a PR which fixes this issue and supports these resolutions.

Thanks,
Max
max@mmaeder.com

XYZ-qiyh, 2024-07-21:
Development Boards: Seeed Studio XIAO ESP32 S3 Sense

MaxMaeder, 2024-09-16:
@XYZ-qiyh What is the XCLK speed you are using? I get `EV-EOF-OVF`

closed 2024-11-23 by github-actions[bot]

Ashley-QwQ, 2026-09-04:
This issue looks closed by the stale bot rather than actually resolved — no PR ever landed here, and the EV-EOF-OVF question from @MaxMaeder never got answered.

I didn't implement set_res_raw for GC2145 (still NULL in this repo), but I did find and verify a working path that doesn't need it: the ordinary set_framesize/camera_config_t.frame_size init path already supports GC2145 up to real 1024×768 (XGA), and 1280×720 (HD) with a lowered XCLK, on real hardware — not just claimed, 100/100 capture success in repeated testing. Full writeup, firmware source, and 100-frame hardware test data (including a boot-time EV-EOF-OVF-style timeout at higher XCLK, which going lower resolved) are here in case it helps anyone else hitting this wall:

https://github.com/Ashley-QwQ/yahboom-esp32s3-gc2145-hires

This is board-specific (2 MB PSRAM Yahboom module), so it's not a drop-in PR for set_res_raw itself — that would still need someone to implement the raw-window path this issue originally asked for. But since the standard frame_size path clearly can reach higher resolutions on GC2145 already, it might be a useful data point for anyone attempting that fix.