espressif/esp32-camera #672: OV2640 standby support

jksemple, 2024-07-30:
This issue revisits https://github.com/espressif/esp32-camera/issues/101 which was closed due to inactivity.
However, @me-no-dev commented that he 'will implement this very soon' - which was 4 years ago.

This is still a problem and effectively prevents ESP32 Camera being used in any low power / battery role.

jksemple, 2024-09-01:
Sorry - I left out obtaining the sensor pointer.  Once you have that it should resolve sensor->set_reg(...) ok :
sensor_t* sensor = esp_camera_sensor_get();

NB the enable parameter enables camera standby mode so set it to false to start the camera up again

wormyrocks, 2024-10-18:
hi, 
I managed to put an OV2640 into standby with the following:
```
// i=1: sleep, i=0: wake
void camera_deep_sleep(int i)
{
  sensor_t *s = esp_camera_sensor_get();
  s->set_reg(s, 0x109, 0x10, i ? 0x10 : 0);
}
```
let me know if you find any way to wake it back up :)

jksemple, 2024-10-18:
Putting it to sleep is not the issue. Getting it to wake up again is where we all get stuck. It appears to be a hardware design flaw. The only workaround is to cut power and restart.

closed 2024-12-25 by github-actions[bot]