espressif/esp32-camera #674: ESP32-S3-MINI-N4R2

jfras, 2024-08-05:
I can run the example code on the ESP32-S3-MINI-N8 version, but only in small resolutions, no jpeg due to the small memory size. I have replaced the module to ESP32-S3-MINI-N4R2 hoping to solve the issue with the additional PSRAM, but seems that even the previous code (with nor without PSRAM usage) does not work on that module. I tried various configurations, the PSRAM itself seems to work when I test it, but the camera does not. As far as I understand, the N8 and N4R2 versions have the same size of DRAM, so that there should be no difference in how they work if the PSRAM is not being used. Is that correct?

My camera configuration:
```
.xclk_freq_hz = 10000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_GRAYSCALE, 
    .frame_size = FRAMESIZE_96X96,    
    .jpeg_quality = 12, 
    .fb_count = 1,      
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
};
```

I am getting:
```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (cam_task) 
Core  0 register dump:
PC      : 0x4037d9e7  PS      : 0x00060a36  A0      : 0x8037b46a  A1      : 0x3fceccc0  
A2      : 0x3fced674  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060a23  
A6      : 0x00060a23  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x3fced628  A11     : 0x00001800  A12     : 0x3fc988cc  A13     : 0x00000000  
A14     : 0x02ced674  A15     : 0x00ffffff  SAR     : 0x00000019  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace: 0x4037d9e4:0x3fceccc0 0x4037b467:0x3fcecd00 0x4037b5de:0x3fcecd20 0x4037b85c:0x3fcecd40 0x403785b4:0x3fcecd60 0x403785e2:0x3fcecd80 0x40378721:0x3fcecdb0 0x4200d93a:0x3fcecdd0 0x4200c981:0x3fcecdf0 0x4200c605:0x3fcece10 0x4201f071:0x3fcece30 0x4201dfbe:0x3fcece50 0x4201e046:0x3fcece70 0x4201eaa1:0x3fcecea0 0x42029dd5:0x3fceced0 0x4202586b:0x3fcecef0 0x420259e1:0x3fced200 0x4202dc1d:0x3fced230 0x4200f9ba:0x3fced260 0x4200fa4f:0x3fced290 0x42010646:0x3fced310

ELF file SHA256: 1cdf04193f446cba
```

jksemple, 2024-08-05:
JPEG takes the least memory as it compresses the image in the camera before it is delivered to your code. Use that rather than grayscale or any other format until you can work out what is going wrong.

jfras, 2024-08-06:
Turns out I needed to re-soldered the ESP module and that the camera itself was faulty:
After re-soldering the module the camera started to capture the grayscale images but was failing to capture JPEG - the `esp_camera_fb_get();` was returning null. Other formats were not working neither. 
Swapping the camera itself solved the JPEG issue. Now everything works. Thank you!

closed 2024-08-06 by jfras