espressif/esp32-camera #676: Incorrect handling of RGB565 images during conversion to BMP or RGB888

jksemple, 2024-08-20:
There is an inconsistency in the handling of RGB565 images in memory.  When decoding a JPG the 16-bit RGB565 pixel value is composed (conversions/to_bmp.c lines 150-155) with R G B bits in that order in the word and then written to memory as low-byte high-byte.  This little-endian order is consistent with the ordering of bytes for other 16-bit or 32-bit values in a BMP e.g. image length.

However, when converting to BMP or RGB888 format the RGB565 pixel value is read as if it were stored as high byte followed by low byte (conversions/to_bmp.c lines 360 and 361 and lines 254 255). This has a psychedelic effect on the colours in resulting images.
A workaround is to swap the order of bytes in an RGB565 image before such a conversion but this is not an acceptable fix.

Please swap lines 360/361 and lines 254/255

jksemple, 2024-10-20:
This is still a problem

cnadler86, 2024-10-25:
I tried this out because I was experiencing some weird issue when changing the frame between conversions and actually swapping the lines as proposed produced by me the psychedelic image. I think the issue might be somewhere else?

jksemple, 2024-10-25:
The problem is that some parts of the library assume the RGB565 order is little-endian and others big-endian. If you convert a JPG image to RGB565 the library conversion makes the byte order little-endian, which then clashes with the BMP conversion code that assumes the order is big-endian.  However, if you read an RGB565 image from your camera and your camera presents the image in big-endian order the conversion to BMP will be fine.  If you tried swapping the bytes in that scenario you would introduce the psychedelic effect as you experienced. My point is simply that the RGB565 byte order matters and needs to be treated consistently throughout the library, which is not the case just now.

cnadler86, 2024-10-25:
Understand. Thanks for clarifying. Then, a possibility would be to pass an (optional?) mask argument in order to specify the byte order of the source image, right?

jksemple, 2024-10-25:
That's a possible workaround but it would still involve iterating through the image and swapping bytes where necessary. I'd rather the camera driver fixed on one byte ordering that works with all conversions. It seems daft for library users to have to swap bytes outside the driver if you want to convert JPG to RGB565 and then to BMP.  I don't mind if the RGB565 word is stored in memory as low-endian or high-endian, I just want it to be consistent.  One further point: It is possible to request RGB565 from camera sensors (e.g. OV3660) in either big-endian or little-endian order, so the sensor driver code will need to be reviewed to ensure they are all sending RGB565 in the right order to allow efficient frame buffering in memory.

closed 2025-01-01 by github-actions[bot]

negagen, 2025-10-01:
Still a problem today