espressif/esp32-camera #679: E (490) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

NagaAvi, 2024-09-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I tried a lot of times but the issue persists, is it my camera faulty? Do I need to get a new one ?

### Actual behavior (suspected bug)

load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
E (334) esp_cor��&VW}���͡� No core dump partition found!
E (334) esp_core_dump_flash: No core dump partition found!
E (490) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105

### Steps to reproduce the behavior

 -

### Project release version

2.3.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

other (details in Additional context)

me-no-dev, 2024-09-13:
Make sure that the pins are correctly configured. Error means it can not connect to your camera's SCCB port

closed 2024-09-13 by me-no-dev

NagaAvi, 2024-09-14:
> Make sure that the pins are correctly configured. Error means it can not connect to your camera's SCCB port

I'm using ESP-32 MB and I've ensured that all the pins working properly. But still the same error.