espressif/esp32-camera #686: calling esp_camera_init() after esp_camera_deinit() generates error message (but continues to work)

vvb333007, 2024-09-22:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

No error messages

### Actual behavior (suspected bug)

It all works. But produces error messages if initialized/deinitialized more than once:
"**E (541244) gpio: gpio_install_isr_service(500): GPIO isr service already installed**"

### Steps to reproduce the behavior

It all works. But produces error messages:

To reproduce:

1. intialize the camera (**esp_camera_init()**) (initialization sequence is taken from esp-camera component's README.md)
2. make some pictures
3. deinit camera (**esp_camera_deinit()**)
4. initialize camera again (**esp_camera_init()**)
5. get error message from ESP-IDF: "**E (541244) gpio: gpio_install_isr_service(500): GPIO isr service already installed**"
6. camera continues to work as expected.

Hardware: Ai Thinker ESPCam (V2640, UXGA mode, JPEG, 2 framebuffers, 20MHz clock)
Framework: Arduino core (used IDF: idf-release_v5.1-b6b4727c58)

### Project release version

idf-release_v5.1-b6b4727c58

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

CMD

vvb333007, 2024-09-22:
Another observation:
after calling to esp_camera_deinit() the camera was powered down (via pin32, PWDN pin). Immediately messages
appeared:

cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF

Looks like _deinit() does not remove isr handler?

me-no-dev, 2024-09-30:
yes. we had to do it that way, because the ISR subsystem might be used by other things in the firmware, so if we shut it off, it would make other things stop working (had such reports from users).