espressif/esp32-camera #693: esp32s3 camera API request

GastonMelo, 2024-10-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

According to TRM pag 1060 the esp32s3 has LCD/camera controller but in the esp32s3 idf programming guide there's only the LCD API's.
Is it possible to get the camera API's?

closed 2024-12-18 by github-actions[bot]

AgainPsychoX, 2024-12-21:
I hate "stale" bots

GastonMelo, 2024-12-21:
Who doesn't

igrr, 2024-12-21:
Hi, sorry that this question went unnoticed.

We have recently added DVP interface support to esp_driver_cam component, but this only works on the ESP32-P4 for the time being. We will add ESP32-S3 support in the future.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-reference/peripherals/camera_driver.html