espressif/esp32-camera #695: About idf_componeAbout

nopnop2002, 2024-10-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Is this definition necessary?

If we use the camera_example directory after copying it to another directory, problems may occur.

https://github.com/espressif/esp32-camera/blob/master/examples/camera_example/main/idf_component.yml#L4

igrr, 2024-10-21:
You are right, with such a sequence of commands you will get a error die to the parent component not being found.

However, we still prefer to keep this definition as such usage doesn't seem common. Currently, two main use cases are supported:

- Component developer use case. Component developer works with a git clone of the repository. They run examples in-place, and expect any modifications done to the components source code be immediately reflected when building the example. Override_path guarantees that we use the local copy of the component, rather than downloading the release from the registry.
- Component user who would like to run the example. Component user can ran `idf.py create-project-from-example` command shown next to the desired example in the component registry. When the example is downloaded, we automatically remove the override_path, thus the example can be built in the absence of the parent component.

nopnop2002, 2024-10-21:
Thank you for replay.
I confirmed that override_path is automatically removed when using idf.py create-project-from-example.

closed 2024-10-21 by nopnop2002