espressif/esp32-camera #696: ESP32 Camera pin connection

sheetaljain01, 2024-10-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, I am trying to make my own PCB with an ESP32-S3. Can I use any GPIO pin for the camera connection, or are there specific recommendations for ensuring good quality?

sheetaljain01, 2024-11-15:
Hi,  

I have used these pins in my custom PCB for the ESP32-S3 camera. Initially, the camera worked fine, but later I started encountering this issue continuously."

#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM     14
#define SIOD_GPIO_NUM     42
#define SIOC_GPIO_NUM     41

#define Y9_GPIO_NUM       21
#define Y8_GPIO_NUM       13
#define Y7_GPIO_NUM       12
#define Y6_GPIO_NUM       10
#define Y5_GPIO_NUM       8
#define Y4_GPIO_NUM       17
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM       9
#define VSYNC_GPIO_NUM    48
#define HREF_GPIO_NUM     47
#define PCLK_GPIO_NUM     11

[b]Arduino Error:[/b]

[b]1)[/b]

E (521) esp_core_dump_flash: No core dump partition found!
E (521) esp_core_dump_flash: No core dump partition found!

E (1064) sccb: SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:263
E (2065) camera: Detected camera not supported.
E (2065) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
E (2065) gdma: gdma_disconnect(299): no peripheral is connected to the channel
Camera init failed with error 0x106

----------------------------------------------------
[b]2)[/b]

E (521) esp_core_dump_flash: No core dump partition found!
E (521) esp_core_dump_flash: No core dump partition found!

E (9028) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (9028) gdma: gdma_disconnect(299): no peripheral is connected to the channel
Camera init failed with error 0x105

----------------------------------------------------
[b]3)[/b]

E (521) esp_core_dump_flash: No core dump partition found!
E (521) esp_core_dump_flash: No core dump partition found!

E (1481) sccb: SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:263
E (3115) sccb: SCCB_Read Failed addr:0x30, reg:0x0a, data:0x00, ret:263
E (3115) camera: Detected camera not supported.
E (3115) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
E (3121) gdma: gdma_disconnect(299): no peripheral is connected to the channel
Camera init failed with error 0x106

skshadan, 2024-12-12:
hi, can i use camera with esp32-wroom-32E?

closed 2025-02-18 by github-actions[bot]