espressif/esp32-camera #706: Camera Initialization Issue on ESP32-S3 Custom PCB

hardikofficial963, 2024-11-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

We are currently developing a custom PCB based on the ESP32-S3 that integrates the following components:
    - OV5640 Camera
    - ES8311 Codec IC (for microphone and speaker functionality)
    - Servo Port
    - I2C Port
    - SD Card Port

While all other components are functioning as expected, we are facing a recurring issue with the camera.

Issue Description:
The PCB works fine initially after assembly, but after some time, the camera fails to initialize, showing errors such as:

    "Camera initialization error"
    "Camera SCCB failed"

Steps Taken:

1. Testing with Custom PCB:
The camera works immediately after assembly but stops functioning after some time.
Testing with ESP32-S3 Development Board:

2. We connected the OV5640 camera using an FFC breakout board and the same pins used in the custom PCB. The same initialization errors occur.

Attachments for Reference:
    - PCB Design Schematic  (1-4, 2-4)
    - PCB Layout   (3-4)
    - List of Pins Used  (4-4)

Could you please help us identify and resolve the issue? Any insights or suggestions would be greatly appreciated.

Looking forward to your guidance.
[image]
[image]
[image]
[image]

06GitHub, 2024-11-30:
Hello,
For my project, I am also implementing ESP32S3 and OV5640 FCC camera on a custom PCB.

Have you tested your PCB with the example Web server provided with Arduino IDE, ie "CameraWebServer"  ? This example, as you probably know, is available at "File ->Examples -> ESP32 -> CameraWebServer"
This Web server is robust and implemented on top of core Espressif APIs.

Imho, if problem no longer occurs, it means that problem is caused by some software bugs in project (I2S ? Web server ? ...). If problem still occurs, it means that problem is caused by PCB.

See example of CameraWebServer snapshot  
[image]

**My configuration**
- Arduino IDE 1.8.19
- arduino-esp32  2.0.17 https://github.com/espressif/arduino-esp32
- ESP32S3 module ESP32-S3-WROOM-1U-N16R8
- camera OV5640 FCC (ribbon, not breakout board)
- Web server PsychicHttp https://github.com/hoeken/PsychicHttp

Hope this helps

hardikofficial963, 2024-11-30:
> Hello, For my project, I am also implementing ESP32S3 and OV5640 FCC camera on a custom PCB.
> 
> Have you tested your PCB with the example Web server provided with Arduino IDE, ie "CameraWebServer" ? This example, as you probably know, is available at "File ->Examples -> ESP32 -> CameraWebServer" This Web server is robust and implemented on top of core Espressif APIs.
> 
> Imho, if problem no longer occurs, it means that problem is caused by some software bugs in project (I2S ? Web server ? ...). If problem still occurs, it means that problem is caused by PCB.
> 
> See example of CameraWebServer snapshot ![Screenshot 2024-11-30 at 10-29-19 ESP32 OV5640](https://private-user-images.githubusercontent.com/11319457/391226204-7d993ae0-0449-47f6-91a5-b0cef76d860e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5NjUzMTUsIm5iZiI6MTczMjk2NTAxNSwicGF0aCI6Ii8xMTMxOTQ1Ny8zOTEyMjYyMDQtN2Q5OTNhZTAtMDQ0OS00N2Y2LTkxYTUtYjBjZWY3NmQ4NjBlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDExMTAxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmNWQwODkyZWM0YTBkNzcxNWMwZTgxN2VkZTAxNmYyZDEzMjg3MzUwNzZiNDUzNzYwYjkyNmMzYmI4NzcxYzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pQ6oiHd_tSn4-IZCwUTuBMg4wnV5t7CXmtf2bU-rcKE)
> 
> **My configuration**
> 
> * Arduino IDE 1.8.19
> * arduino-esp32  2.0.17 https://github.com/espressif/arduino-esp32
> * ESP32S3 module ESP32-S3-WROOM-1U-N16R8
> * camera OV5640 FCC (ribbon, not breakout board)
> * Web server PsychicHttp https://github.com/hoeken/PsychicHttp
> 
> Hope this helps

@06GitHub 
Yes, I have tested Camera Webserver Example on Arduino, Not working.

Which Pins you are using in your custom PCB ? If possible can you share Pins and PCB layout ?

Do you find any mistake in my Camera Pin Routing ?

06GitHub, 2024-11-30:
**1) Camera schematics** 
[image]

**2) Camera pinout**
[image]

**3) Power supplies**
Custom PCB implementation requires specific power supplies and dedicated resistors/capacitors, as per OV5640 data sheet.
Can you check that power supplies are OK ? 2.8v, 1.2 or 1.5v, 3.3v
(check can be done with a multimeter or an oscilloscope)

**4) XCLK**
On my project I use an 24MHz oscillator to provide clock signal XCLK to OV5640 FCC camera (clock is not provided by ESP32). Can you check that on your project XCLK is ok (should be done with an oscilloscope) ?

Those 2 problems were encountered on my project, some months ago.

**5) SDA, SCL pull up**
Pull up resistors (10K) should be present on SDA and SCL pins.

**6) Test connectors**
Connectors "CAMERA BREAKOUT BOARD" and "CAMERA TEST" are present for test/debugging purpose.

MedKhalil19, 2025-04-10:
Hi everyone!
I'm planning to design a custom board using the ESP32-S3, OV2640 camera, and an SD card. I'm still new to hardware design, so I’d really appreciate it if you could share a reference schematic or design example that I can learn from and use as a guideline to avoid mistakes.
Thanks a lot in advance!

06GitHub, 2025-04-12:
Hello MedKhalil19,
OV5640 camera custom PCB above works for me with ESP32S3. I guess that OV2640 should also be OK (I started my project with OV2640 and switched to OV5640 since I find OV5640 more reliable).

This PCB allows to connect 2 types of camera :
- Camera Breakout Board, optional, for debugging if video with Camera FCC 24P (ribbon) fails
[image]
- Camera FCC 24P (ribbon)
[image]

One important point is that Camera FCC 24P (ribbon) needs : 
- 2 additional power supplies (1.2V and 2.8 V) in addition to 3.3V
- 1 external 24MHz clock (oscillator)

This is not needed for Camera Breakout Board since it is embedded on breakout board.

PCB includes test connector, useful to test signals with an oscilloscope if video fails.

Code to do video streaming with PCB, PsychicHttp server and ESP32S3 is here https://github.com/hoeken/PsychicHttp/discussions/143#discussioncomment-10299770 .

[image]

For SD Card, schema below shows how to use an SD card connector with SPI interface, such as :

[image]

https://www.lcsc.com/product-detail/SD-Card-Memory-Card-Connector_XUNPU-TF-115K_C266621.html

[image]

if you are using Arduino IDE, one can find many examples in menu "File->Examples->SD or SD_MMC" how to use/connect an SD card (both for SPI and SD_MMC interfaces).
RandomnerdTutorials provides also useful information https://randomnerdtutorials.com/esp32-microsd-card-arduino/

06GitHub, 2025-04-13:
Hello,

`I'm using an ESP32-S3, am I correct in thinking I can generate the XCLK (24 MHz) using a PWM output from the ESP32-S3, or alternatively use an external oscillator?`
Indeed theoritically one could use an ESP32S3 PWM output to generate a 24Mhz signal for XCLK. For my project, I preferred to rely on an external clock (24 Mhz oscillator) to keep ESP32S3 "busy" with other tasks (video streaming, audio ...). You could test your PCB 1) first with a ESP32S3 PWM 24Mhz signal and if not satisfactory, 2) switch to external oscillator 24Mhz.

`I'm not entirely sure about the function of the CD/DATA3 pin in your setup, I believe it might be optional. Could you clarify that?`
CD/DATA3 is not optional, it is **mandatory** since it corresponds to **CS** signal for SPI interface :

SPI interface relies on 4 signals :
CS
MOSI
CLK
MISO
 
As you probably know, it is also possible to use SD_MMC interface instead of SPI interface. In that case, the schema has minor changes.

`Also, if you don’t mind, could you share a way to contact you? I’d love to discuss this further with you. Thanks!`
You may contact me on my email, or you can also use this issue (to share information)

Hope this helps

MedKhalil19, 2025-04-13:
Hi,

For the 24 MHz signal: I’ll start with the ESP32-S3’s PWM output. Since I’m not using all the pins, I’d prefer to avoid adding extra components at this stage. But as you mentioned, if the signal turns out to be unstable or not precise enough, I’ll switch to an external 24 MHz oscillator.

Regarding the CD/DATA3 pin: yes, that’s exactly what I meant when i said, CD/DATA3 might be optional in SPI mode, it’s mandatory because it acts as the CS signal. I haven’t tried SD/MMC mode yet, but from what I’ve read, it’s faster than SPI, though more complex to implement.

Btw Do you have any ideas or advice on designing a custom ESP32-based board that’s battery-powered, with charging, protection, and the ability to switch between a 5V input and a 2S lithium battery?

Also, I’d love to hear more about your custom board, what’s its purpose, did you fully build and test it, and did you run into any issues?

Thanks again for your help!

06GitHub, 2025-04-15:
For charging a lithium battery, it is possible to use tp5000 charger module (breakout board version shown below, exists also in other versions such as Dual Inline Package DIP that can be integrated in a PCB, ...)

[image]

If your project is battery powered, it could be useful to monitor current consumption. "Nordic Power Profiler II" is a good tool for this 

[image]

For instance, you will notice that when Wifi is on, ESP32S3 current consumption is (much) higher.

MedKhalil19, 2025-05-03:
Yes, I think this will be very helpful. Thank you for your advice and great ideas!