espressif/esp32-camera #713: Uses old I2C driver. Please update to use i2c_master.h

codewitch-honey-crisis, 2025-01-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect to get no warnings on startup, and to be able to use the recommended I2C drivers with my I2C devices.

### Actual behavior (suspected bug)

The camera driver uses legacy i2c drivers, causing a warning on startup as well as limiting me to using legacy I2C drivers for I2C coms.

### Error logs or terminal output

```shell
W (779) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
```

### Steps to reproduce the behavior

1. Include the camera component while targeting the latest ESP-IDF 5.x

### Project release version

2.0.15

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

me-no-dev, 2025-01-31:
The new driver is added : https://github.com/espressif/esp32-camera/blob/master/driver/sccb-ng.c
It's enabled on IDF 5.4 and up: https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L89

Are you sure you are using the latest version of this driver?

codewitch-honey-crisis, 2025-01-31:
I don't remember now. It's possible I was accidentally on an older version. Sorry about that.

Hailaylin, 2025-04-02:
> The new driver is added : https://github.com/espressif/esp32-camera/blob/master/driver/sccb-ng.c新驱动程序已添加：https://github.com/espressif/esp32-camera/blob/master/driver/sccb-ng.c It's enabled on IDF 5.4 and up: https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L89它在 IDF 5.4 及更高版本上启用：https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L89
> 
> Are you sure you are using the latest version of this driver?您确定您使用的是此驱动程序的最新版本吗？

when i use new driver, it print error
```
I (763) sccb-ng: pin_sda 3 pin_scl 4
I (766) sccb-ng: sccb_i2c_port=1
I (770) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (797) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (827) camera: Detected camera at address=0x21
E (828) i2c.master: I2C transaction unexpected nack detected
E (828) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (832) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (839) sccb-ng: SCCB_Read Failed addr:0x21, reg:0x0a, data:0x79, ret:259
I (845) ov7725: Mismatch PID=0x79
E (848) i2c.master: I2C transaction unexpected nack detected
E (853) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (860) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (867) sccb-ng: SCCB_Write Failed addr:0x21, reg:0xff, data:0x01, ret:259
E (874) i2c.master: I2C transaction unexpected nack detected
E (879) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (886) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (893) sccb-ng: SCCB_Read Failed addr:0x21, reg:0x0a, data:0x79, ret:259
I (899) ov7670: Mismatch PID=0x79
E (902) i2c.master: I2C transaction unexpected nack detected
E (908) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (914) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (921) sccb-ng: SCCB_Read Failed addr:0x21, reg:0xf1, data:0x79, ret:259
E (928) i2c.master: I2C transaction unexpected nack detected
E (933) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (940) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (947) sccb-ng: SCCB_Read Failed addr:0x21, reg:0xf0, data:0x79, ret:259
I (953) gc032a: Mismatch PID=0x7979
E (956) i2c.master: I2C transaction unexpected nack detected
E (961) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (968) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (975) sccb-ng: SCCB_Write Failed addr:0x21, reg:0xfe, data:0x00, ret:259
E (982) i2c.master: I2C transaction unexpected nack detected
E (987) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (994) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (1001) sccb-ng: SCCB_Read Failed addr:0x21, reg:0x00, data:0x79, ret:259
E (994) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (994) i2c.master: i2c_master_transmit_receive(1220): I2C transaction failed
E (1001) sccb-ng: SCCB_Read Failed addr:0x21, reg:0x00, data:0x79, ret:259
I (1007) gc0308: Mismatch PID=0x79
E (1010) camera: Detected camera not supported.
E (1015) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
E (1021) gdma: gdma_disconnect(319): no peripheral is connected to the channel
E (1028) example:take_picture: Camera Init Failed
I (1032) main_task: Returned from app_main()
```
i guess new i2c driver care about nack, but sccb have not

AxelLin, 2025-05-21:
> The new driver is added : https://github.com/espressif/esp32-camera/blob/master/driver/sccb-ng.c It's enabled on IDF 5.4 and up: https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L89
> 

@me-no-dev 

But enable new driver on IDF 5.4 causes regression because the new i2c drvier
does not co-exist with old driver. (Some libraries still use old i2c driver).
ps. The point is we cannot expect when the 3rd libraries convert to new i2c driver.

IMHO, as long as the IDF still supports old driver, the user should be able to use it.
i.e. You need to allow build with old driver in IDF 5.x.

Also see: https://github.com/espressif/esp32-camera/issues/741

closed 2025-07-27 by github-actions[bot]