espressif/esp32-camera #715: Enroll face button and sliders missing

GramOfCarbs, 2025-01-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

On the camerawebserver example for the esp32 cam, the IP address it gives me used to have the "enroll face" button and the "face detection"/"face identification" button but... they're all missing. I looked through my code and couldn't find any errors. Any idea as to why this happened?

[image]

me-no-dev, 2025-01-31:
Starting with Arduino 3.1, face detection and recognition is not supported on ESP32. You can use 3.0.7 instead. There have been some updates to the IDF component that provides that functionality, but we have not had time to look at what would be now possible and if it is again supported in any way.

nehashetty7, 2025-02-18:
I'm also facing the same could you please give the troubleshooting method for it

closed 2025-04-27 by github-actions[bot]

kausht, 2025-05-21:
hey i also have the same issue please help