espressif/esp32-camera #720: Image capture problem

4324250, 2025-01-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hello, my problem is that when displaying the image captured by ov5640 it does not show a valid image, I tried several configurations and it always returns the same image.

camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D9,
    .pin_d6 = CAM_PIN_D8,
    .pin_d5 = CAM_PIN_D7,
    .pin_d4 = CAM_PIN_D6,
    .pin_d3 = CAM_PIN_D5,
    .pin_d2 = CAM_PIN_D4,
    .pin_d0 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D3,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    // XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 16000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RGB565, // YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_96X96,    // QQVGA-UXGA, For ESP32, do not use sizes above QVGA when not JPEG. The performance of the ESP32-S series has improved a lot, but JPEG mode always gives better frame rates.

    .jpeg_quality = 12, // 0-63, for OV series camera sensors, lower number means higher quality
    .fb_count = 2,      // When jpeg mode is used, if fb_count more than one, the driver will work in continuous mode.
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_LATEST,

[image]

### Actual behavior (suspected bug)

..

### Error logs or terminal output

```shell
..
```

### Steps to reproduce the behavior

..

### Project release version

utilmo

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

debian

### Shell

ZSH

me-no-dev, 2025-01-31:
Which chip are you using?

4324250, 2025-02-03:
esp32s3 r8n8

the idea is to take a photo using psram and then send it over the
network. probe rgb565 takes the capture but it looks bad and when I
use jpeg it returns the buffer to NULL

thank you.

me-no-dev, 2025-02-05:
try with XCLK 20MHz, not 16MHz

closed 2025-02-11 by 4324250

4324250, 2025-02-14:
[image]
The other problem I have is that I capture a solid image of only one color and it captures less than half of the image and then fills it with garbage. If you have any information that can help me solve this. Thank you.
It is in RGB format in JPEG and I always get frame null.