espressif/esp32-camera #721: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

elharamye, 2025-02-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, im a beginner with arduino and these type of things but my esp32 cam [LINK](https://www.amazon.it/diymore-Bluetooth-ESP32-CAM-MB-Dual-core-Fotocamera/dp/B08X3GRK22/ref=sr_1_5?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3CZ2P4HFDN2LP&dib=eyJ2IjoiMSJ9.2fVZO3KfWVdehb33X-ymAY6ixk-h37Jae97lrL2Xv_BqjjgDJ08lS7zUhkR6gAwCac51wCMBClXDeovaXBaE3qluboCdJBgnU1ADQe1CwipLy_oPdIhZlwXtibVYJycszR3WGHgnT7pVVjjOGiihLtCSRLAkJ7I_bKvbVUmgux0B0kYEG4a5gi8xHvtpKL-QCWNJLCc2FUqFRza5Pgh30IVs7M9yBM_6AhfaGMQ4DnZZbTXnaP5CBGh24-QMF7VPRfqOYMut6xCCxEEMV0nXUmQVfEdjCEmh8xazfpFfUPbMD6Ufbj5MuweaZpWJiZZbLECsWzD9nVUcseFRRd2FDe6uMRFd2aUqVOp3piHArDtkNTQPB3HkYSh6_qFTf3tM0sG2IA0B6TW62tE0mZ5gx1vljx5FD1460T1J0YHQwhjTXr-5AFVHYU75NEz6UFCK.pBcz03d9dsha4092r0kM_cuoPFoWSmHzh2y2vNLO7Jg&dib_tag=se&keywords=esp32%2Bcam&qid=1738446862&sprefix=esp32%2Bcam%2Caps%2C124&sr=8-5&th=1)

Get this error, someone knows why and how to fix it? I already saw an issue closed but i didnt understood much since i am a beginner

me-no-dev, 2025-02-05:
You have not selected the correct pins for your camera. There are many tutorials for that board online, but also there are many problems with it

elharamye, 2025-02-05:
thank you for the answer but, in the amazon link you can see that he has like an automatic thing to attach the esp32 and the pc and i cant see what pins are

elharamye, 2025-02-07:
> You can ask the seller for a schematic or data manual

thank you

closed 2025-04-16 by github-actions[bot]