espressif/esp32-camera #722: ask for module of SC031GS

wwsinsh, 2025-02-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

@WangYuxin-esp 您好，我目前在使用esp32S2的开发板，我想购买一个SC031GS模块来做相机开发，想问下您移植驱动时使用的是哪个，可否提供下购买链接

WangYuxin-esp, 2025-02-13:
尽量选 ESP32S3 吧。相机模块要自己到网上采购。

wwsinsh, 2025-02-20:
选型已经定了，我这边想为ESP32-camera新增OV9281的驱动，但是ESP32的文档并没有相关的流程介绍，这个有具体规范和验证流程吗

closed 2025-02-20 by wwsinsh

WangYuxin-esp, 2025-02-20:
> 选型已经定了，我这边想为ESP32-camera新增OV9281的驱动，但是ESP32的文档并没有相关的流程介绍，这个有具体规范和验证流程吗
OV9281 是一个RAW sensor，ESP32S3 没有 ISP，没有自动曝光控制，降噪处理。

wwsinsh, 2025-02-25:
我这边只需要拿到raw就可以，我自己移植了一下，但是DMA接收不到数据，cam_hal: FB-SIZE: 0 != 921600

WangYuxin-esp, 2025-02-25:
> 我这边只需要拿到raw就可以，我自己移植了一下，但是DMA接收不到数据，cam_hal: FB-SIZE: 0 != 921600

应是 sensor 没有起来。sensor 与开发板的硬件接口不匹配。

wwsinsh, 2025-02-25:
我量了VSYNC/HSYNC/PCLK/DATA这些信号应该没问题呢

wwsinsh, 2025-02-25:
目前看是没有DMA中断，没有搬运数据