espressif/esp32-camera #728: support for OV9281 on ESP32S2, error: cam_hal: FB-SIZE: 0 != 921600

wwsinsh, 2025-02-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

receive correct data

### Actual behavior (suspected bug)

no DMA data

### Error logs or terminal output

```shell
I (3615) ESP32-OV9281: i2c master init
I (3615) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (3615) cam_hal: cam init ok
W (3615) ledc: GPIO 1 is not usable, maybe conflict with others
I (3615) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3635) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3665) camera: Detected camera at address=0x60
I (3665) camera: Detected OV9281 camera
I (3665) camera: Camera PID=0x9281 VER=0x00 MIDL=0x00 MIDH=0x00
I (3675) ov9281: reset
I (3905) s2 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3
I (3905) s2 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 12, dma_buffer_size: 30720
I (3905) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 60
I (3915) cam_hal: Allocating 921600 Byte frame buffer in PSRAM
I (3975) cam_hal: Allocating 921600 Byte frame buffer in PSRAM
I (4035) cam_hal: cam config ok
I (4035) ov9281: Set framesize: 13
I (4035) ESP32-OV9281: Camera Init Success
E (4055) cam_hal: FB-SIZE: 0 != 921600
E (4075) cam_hal: FB-SIZE: 0 != 921600
E (4085) cam_hal: FB-SIZE: 0 != 921600
E (4095) cam_hal: FB-SIZE: 0 != 921600
```

### Steps to reproduce the behavior

1. I migrate the 9281 driver for esp-camera 
2. pclk->26Mhz

### Project release version

v2.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

ZSH

### Additional context

there is no valid data in DMA while vsync intr, i want to know what actions would cause this phenomenon

huyuchen0824, 2025-02-25:
hello, I'm working on migration of OV9281 to ESP32S3 either. my mail address: 3195016438@qq.com. Wish to discuss with U :)

wwsinsh, 2025-02-26:
+你QQ了

asadm, 2025-03-25:
hello, did you figure this out?

wwsinsh, 2025-04-14:
> hello, did you figure this out?

increase the HTS and VTS timming, especially HTS. There should be enough free time left to handle DMA data for the ESP32

closed 2025-04-14 by wwsinsh