espressif/esp32-camera #735: esp_camera_fb_get获取的不是最新的一帧，而是最新一帧的上一帧图像？

zhoumiaomiao, 2025-04-08:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

条件：使用sc031gs trigger mode，当trigger信号发起之后，调用esp_camera_fb_get()可以获取该trigger时刻的图像

### Actual behavior (suspected bug)

条件：使用sc031gs trigger mode，当trigger信号发起之后，按理说调用esp_camera_fb_get()可以获取trigger时刻的图像，但是发现获取到的图像是上一次trigger时刻的图像。
主控模组：esp32s3
第一次发起trigger之后，调用esp_camera_fb_get()没有图像，第二次发起trigger之后，调用esp_camera_fb_get可以获取图像，但是是第一次trigger时刻的图像，第三次发起trigger之后，调用esp_camera_fb_get可以获取图像，但是是第二次trigger时刻的图像..........esp_camera_fb_get获取到的永远是上一次trigger的图像

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

主控模组：esp32s3  使用sc031gs trigger mode
camera配置如下：
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,
    //XCLK 6MHz for sc031gs
    .xclk_freq_hz = 6000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,
    .pixel_format = PIXFORMAT_GRAYSCALE, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_VGA,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG
    // .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 3,       //if more than one, i2s runs in continuous mode. Use only with JPEG
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_LATEST,
};
1、第一次发起trigger之后，调用esp_camera_fb_get()没有图像；
2、第二次发起trigger之后，调用esp_camera_fb_get可以获取图像，但是是第一次trigger时刻的图像；
3、第三次发起trigger之后，调用esp_camera_fb_get可以获取图像，但是是第二次trigger时刻的图像..........
esp_camera_fb_get获取到的永远是上一次trigger的图像

### Project release version

esp-idf-v5.2.1   

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows11

### Shell

CMD

WangYuxin-esp, 2025-04-21:
可以调过旧帧，开获取最新的帧，参考 [33](https://docs.espressif.com/projects/esp-faq/zh_CN/latest/application-solution/camera-application.html#esp32-camera-capture)、[34](https://docs.espressif.com/projects/esp-faq/zh_CN/latest/application-solution/camera-application.html#esp32-camera-sdk)

zhoumiaomiao, 2025-04-21:
我测试过了，跳帧是可以实现延时一帧的问题。但是我遇到的trigger mode下的问题是最新trigger的一帧通过esp_camera_fb_get接口获取不到数据（返回的报错是：cam_hal: Failed to get  the frame on time!   说明当前frame_buff里面没有数据），直到下一次trigger来了之后，上一帧数据才能通过esp_camera_fb_get获取到。所以通过跳帧也获取不到最新的一帧。咨询一下，dvp的数据通过dma到frame_buffer的过程是实时的嘛？还是说得等下一帧数据过来，该帧才会塞进frame_buff中？

WangYuxin-esp, 2025-04-21:
camera sensor 的数据是不间断发送到 SOC 的，但是 SOC 的 frame buffer 不是无限多的。  
框架提供了两种缓存机制，参考上述 [33](https://docs.espressif.com/projects/esp-faq/zh_CN/latest/application-solution/camera-application.html#esp32-camera-capture)，你应该选择 CAMERA_GRAB_LATEST 模式。并且配置 frame buffer 的个数不小于 2 个，并且每次使用 fb_get 后，及时 fb_return。  
如果还慢，那是你的  sensor 配置或者硬件问题。即 sensor 出数据就慢。这点可以使用[示例](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/test_framerate)来测试帧率。

zhoumiaomiao, 2025-04-22:
streaming mode下测试过帧率是16Hz。trigger mode下的帧率就取决于trigger的频率了，现在trigger频率很低(秒级别)，我就是trigger一次去获取一次帧率，获取之后会及时fb_return，然后再trigger下一帧。
camera_config_t配置里面fb_count=3,grab_mode设置的是CAMERA_GRAB_LATEST

zhoumiaomiao, 2025-04-22:
trigger mode下camera的数据不会不间断的发送到SOC的，只有trigger一次才会发送数据到SOC

WangYuxin-esp, 2025-04-22:
那你使用逻辑分析仪抓线吧，理论上 trigger 信号一下，就仅有一个 VSYNC。然后程序中在 cam_task 中加少许的 log，看下获取的是否是这次 trigger 的数据。

zhoumiaomiao, 2025-04-22:
好，vsync我用示波器看过的，trigger一次能测到一个vsync的。那我在cam_task加一些log看看

WangYuxin-esp, 2025-04-22:
framebuffer 要准备好，并且是空的。这样 trigger 后，触发 VSYNC，填充新数据到 framebuffer。   
如果能确保上述流畅，那么就是 sensor 这边的问题了，它是否是一直触发 VSYNC，需要分析下线上信号。作为接收端，没有太多能调的。  
在应用层的启示是，你想要获取最新的帧，那么设置 framebuffer count = 1，fb_get 后，不执行 fb_return，此时就没有可用的 framebuffer；仅在 trigger前，fb_teturn 一下，激活可用的 framebuffer，然后立即执行 trigger.

closed 2025-06-23 by zhoumiaomiao