espressif/esp32-camera #737: ov7670摄像头通讯问题

zyl2995230, 2025-04-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

esp32-idf版本: 5.4
esp32-camera版本: 2.0.15

ov7670摄像头sccb通讯一切正常，但是在PID获取时从机无应答(NACK)，我查看并修改了源码后PID获取成功，摄像头初始化成功。这是什么原因导致的？修改代码如下：

路径：managed_components\espressif__esp32-camera\driver\sccb.c

uint8_t SCCB_Read(uint8_t slv_addr, uint8_t reg)
{
    i2c_master_dev_handle_t dev_handle = *(get_handle_from_address(slv_addr));

    uint8_t tx_buffer[1];
    uint8_t rx_buffer[1];

    tx_buffer[0] = reg;
    i2c_master_transmit(dev_handle, tx_buffer,1,TIMEOUT_MS);
    esp_err_t ret = i2c_master_receive(dev_handle,rx_buffer, 1, TIMEOUT_MS);
    // esp_err_t ret = i2c_master_transmit_receive(dev_handle, tx_buffer, 1, rx_buffer, 1, TIMEOUT_MS);

    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "SCCB_Read Failed addr:0x%02x, reg:0x%02x, data:0x%02x, ret:%d", slv_addr, reg, rx_buffer[0], ret);
    }

    return rx_buffer[0];
}

zyl2995230, 2025-04-14:
示波器检查了，发现PID请求发送后从机应答了，但是在发送PID读取时读写位是低电平，正常来说读数据时为高电平才对

crushonme, 2025-05-30:
哈哈 连OV7725也没搞通，不知道咋回事儿。

zyl2995230, 2025-06-25:
@14790897 你的i2c线路接上拉电阻了吗？

14790897, 2025-06-26:
> > > [@14790897](https://github.com/14790897) 你的i2c线路接上拉电阻了吗？
> > 
> > 
> > 用的是嘉立创的开发板，它里面应该有吧。
> 
> 最好是看看原理图确认一下

https://oshwhub.com/li-chuang-kai-fa-ban/li-chuang-esp32s3r8n8-kai-fa-ban

就这块开发板，我不会看原理图，大佬帮我看一下吧

zyl2995230, 2025-06-26:
> > > > [@14790897](https://github.com/14790897) 你的i2c线路接上拉电阻了吗？
> > > 
> > > 
> > > 用的是嘉立创的开发板，它里面应该有吧。
> > 
> > 
> > 最好是看看原理图确认一下
> 
> https://oshwhub.com/li-chuang-kai-fa-ban/li-chuang-esp32s3r8n8-kai-fa-ban
> 
> 就这块开发板，我不会看原理图，大佬帮我看一下吧

你这个板子i2c没有接上拉电阻

zyl2995230, 2025-07-01:
> > https://github.com/14790897/CameraInLCD/blob/main/main/dvp_lcd_main.c
> > 能读取了，但是显得出来是花屏的,, 求助 视频: https://www.bilibili.com/video/BV1fCK2zcE3Y/
> 
> 我解决了，换了个摄像头，并且加了上拉电阻
> 
> https://www.bilibili.com/video/BV1zwK2zFEBt/

你需要去看摄像头的手册进行初始化和优化！

ACaiAa, 2025-07-01:
> 然后改了大佬的代码之后，现在可以初始化，但是会报这个错:
> 
> ```
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> cam_hal: EV-VSYNC-OVF
> 
> W (1344) cam_hal: FB-OVF
> W (1354) cam_hal: FB-OVF
> ```

降低分辨率好了