espressif/esp32-camera #769: M5 "5MP" does not read images / crashes

RubenKelevra, 2025-07-15:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The M5 "5MP" unit should be supported.

### Actual behavior (suspected bug)

@turenkomv reported [issues](https://github.com/esphome/esphome/pull/8832#issuecomment-2897275655).

### Error logs or terminal output

```shell
Without stack size increase:

 
[19:51:57]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[19:51:57]Debug exception reason: Stack canary watchpoint triggered (framebuffer_tas) 
[19:51:57]Core  1 register dump:
[19:51:57]PC      : 0x4037ef9e  PS      : 0x00060836  A0      : 0x820494aa  A1      : 0x3fcb4780
WARNING Decoded 0x4037ef9e: xPortSetInterruptMaskFromISR at C:\Users\turen\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:552
 (inlined by) xPortEnterCriticalTimeout at C:\Users\turen\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:478
[19:51:57]A2      : 0x3fc9ae6c  A3      : 0xffffffff  A4      : 0x00000007  A5      : 0x3fcb47be  
[19:51:57]A6      : 0x00000066  A7      : 0x00000064  A8      : 0x00060820  A9      : 0x3fcb4780
[19:51:57]A10     : 0x00060823  A11     : 0x00000000  A12     : 0x00060820  A13     : 0x00060223
[19:51:57]A14     : 0x00000000  A15     : 0x0000abab  SAR     : 0x00000018  EXCCAUSE: 0x00000001
[19:51:57]EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff

[19:51:57]Backtrace: 0x4037ef9b:0x3fcb4780 0x420494a7:0x3fcb47b0 0x420495cd:0x3fcb47d0 0x4200f9ed:0x3fcb47f0 0x42016ada:0x3fcb48d0 0x420a5a61:0x3fcb4910 0x40384606:0x3fcb4940 0x4201a84a:0x3fcb4990 0x42019d6a:0x3fcb49d0 0x4200bc76:0x3fcb49f0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4037ef9b: xPortSetInterruptMaskFromISR at C:\Users\turen\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:552
 (inlined by) xPortEnterCriticalTimeout at C:\Users\turen\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:478
WARNING Decoded 0x420494a7: vPortEnterCritical at C:\Users\turen\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:567
 (inlined by) find_key at C:\Users\turen\.platformio\packages\framework-espidf\components\pthread/pthread_local_storage.c:77
WARNING Decoded 0x420495cd: pthread_setspecific at C:\Users\turen\.platformio\packages\framework-espidf\components\pthread/pthread_local_storage.c:199
WARNING Decoded 0x4200f9ed: esphome::logger::Logger::check_and_set_task_log_recursion_(bool) at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/src/esphome/components/logger/logger.h:298
 (inlined by) esphome::logger::Logger::check_and_set_task_log_recursion_(bool) at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/src/esphome/components/logger/logger.h:287    
 (inlined by) esphome::logger::Logger::log_vprintf_(unsigned char, char const*, int, char const*, __va_list_tag) at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/src/esphome/components/logger/logger.cpp:35
WARNING Decoded 0x42016ada: esphome::esp_idf_log_vprintf_(char const*, __va_list_tag) at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/src/esphome/core/log.cpp:56
WARNING Decoded 0x420a5a61: esp_log_writev at C:\Users\turen\.platformio\packages\framework-espidf\components\log/log.c:210
WARNING Decoded 0x40384606: esp_log_write at C:\Users\turen\.platformio\packages\framework-espidf\components\log/log.c:220
WARNING Decoded 0x4201a84a: cam_take at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/managed_components/espressif__esp32-camera/driver/cam_hal.c:541 (discriminator 1)
WARNING Decoded 0x42019d6a: esp_camera_fb_get at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/managed_components/espressif__esp32-camera/driver/esp_camera.c:384
WARNING Decoded 0x4200bc76: esphome::esp32_camera::ESP32Camera::framebuffer_task(void*) at C:\Projects\m5stack-camera-1\.esphome\build\m5stack-camera-1/src/esphome/components/esp32_camera/esp32_camera.cpp:412

[Source](https://github.com/espressif/esp32-camera/pull/765#issuecomment-3045941004)

With stack size increase:

[23:40:49][W][esp32_camera:204]: Got invalid frame from camera!
[23:40:49]cam_hal: EV-EOF-OVF
...
[23:40:51]cam_hal: EV-EOF-OVF
[23:40:51][D][esp-idf:000][cam_task]: W (12646) cam_hal: NO-SOI - 100 additional misses
[23:40:51]cam_hal: EV-EOF-OVF
...
[23:40:53]cam_hal: EV-EOF-OVF
[23:40:53][W][esp32_camera:204]: Got invalid frame from camera!
[23:40:53][D][esp-idf:000][framebuffer_tas]: W (14666) cam_hal: Failed to get frame: timeout
[23:40:53]cam_hal: EV-EOF-OVF
...
[23:40:57]cam_hal: EV-EOF-OVF
[23:40:57][W][esp32_camera:204]: Got invalid frame from camera!
[23:40:57][D][esp-idf:000][framebuffer_tas]: W (18680) cam_hal: Failed to get frame: timeout
[23:40:57]cam_hal: EV-EOF-OVF
...
[23:41:01]cam_hal: EV-EOF-OVF
[23:41:01][W][esp32_camera:204]: Got invalid frame from camera!
[23:41:01][D][esp-idf:000][framebuffer_tas]: W (22693) cam_hal: Failed to get frame: timeout
[23:41:01]cam_hal: EV-EOF-OVF
...
[23:41:01]cam_hal: EV-EOF-OVF
[23:41:01][D][esp-idf:000][cam_task]: W (22934) cam_hal: NO-SOI - 100 additional misses
[23:41:01]cam_hal: EV-EOF-OVF
...
[23:41:05]cam_hal: EV-EOF-OVF
[23:41:05][W][esp32_camera:204]: Got invalid frame from camera!
[23:41:05][D][esp-idf:000][framebuffer_tas]: W (26707) cam_hal: Failed to get frame: timeout
[23:41:05]cam_hal: EV-EOF-OVF
...
[23:41:09]cam_hal: EV-EOF-OVF
[23:41:09][W][esp32_camera:204]: Got invalid frame from camera!
[23:41:09][D][esp-idf:000][framebuffer_tas]: W (30720) cam_hal: Failed to get frame: timeout
[23:41:09]cam_hal: EV-EOF-OVF
...
[23:41:11]cam_hal: EV-EOF-OVF
[23:41:11][D][esp-idf:000][cam_task]: W (32936) cam_hal: NO-SOI - 100 additional misses
[23:41:11]cam_hal: EV-EOF-OVF
...
[23:41:13]cam_hal: EV-EOF-OVF
[23:41:13][W][esp32_camera:204]: Got invalid frame from camera!
[23:41:13][D][esp-idf:000]cam_hal: EV-EOF-OVF
[23:41:13][framebuffer_tas]: W (34731) cam_hal: Failed to get frame: timeout

[Source](https://github.com/espressif/esp32-camera/pull/761#issuecomment-3040155353)
```

### Steps to reproduce the behavior

Setup M5 "5MP" with something like ESPHome and see the esp32-camera library throw errors:

### Project release version

2.1.0-ish

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

unknown

### Shell

Bash

RubenKelevra, 2025-07-15:
@turenkomv The messages in your log suggests that there seems to be a JPEG found, but not at the offset 0. That's why it's not printing an error message not finding the start of the image (`NO-SOI - JPEG start marker missing`), because it found it, but if it's not at `0` the frame it then silently skipped.

I'll create a PR.

RubenKelevra, 2025-07-15:
Okay, thanks! 

I think I found another bug, while investigating the one above. Both could lead to the suggested behavior (I think). But I'm a bit too tired to be sure.

I'll flash the UserDemo tomorrow on my device (which looks identical) to see if it maybe changes anything, if I then switch back to ESPHome.

RubenKelevra, 2025-08-03:
I was unable to establish what's causing the issues you're running into, @turenkomv.

Would you be so kind and share the UserDemo image you used before, privately (E-Mail address is in my profile), so I can flash it and see if this triggers any bug if I flash ESPHome afterwards?

And can you confirm, that the current version of ESPHome still has issues, if you update the ESP32-Camera library to master plus #781?

Here's a patch for ESPHome to fetch the current master plus #781 from the PR branch:

```
From ecfdfcbb535ea77a4b46705186a2c032cb417c49 Mon Sep 17 00:00:00 2001
From: RubenKelevra <user@example.com>
Date: Sun, 3 Aug 2025 16:02:37 +0200
Subject: [PATCH] update esp32-camera lib to master+PR781

---
 esphome/components/esp32_camera/__init__.py | 6 +++++-
 esphome/idf_component.yml                   | 3 ++-
 2 files changed, 7 insertions(+), 2 deletions(-)

diff --git a/esphome/components/esp32_camera/__init__.py b/esphome/components/esp32_camera/__init__.py
index a99ec3408..2a4085c29 100644
--- a/esphome/components/esp32_camera/__init__.py
+++ b/esphome/components/esp32_camera/__init__.py
@@ -333,7 +333,11 @@ async def to_code(config):
     cg.add_define("USE_CAMERA")
 
     if CORE.using_esp_idf:
-        add_idf_component(name="espressif/esp32-camera", ref="2.0.15")
+        add_idf_component(
+            name="esp32-camera",
+            repo="https://github.com/RubenKelevra/espressif_esp32-camera.git",
+            ref="2368d94-PR781",
+        )
 
     for conf in config.get(CONF_ON_STREAM_START, []):
         trigger = cg.new_Pvariable(conf[CONF_TRIGGER_ID], var)
diff --git a/esphome/idf_component.yml b/esphome/idf_component.yml
index c43b62268..186e97304 100644
--- a/esphome/idf_component.yml
+++ b/esphome/idf_component.yml
@@ -2,7 +2,8 @@ dependencies:
   espressif/esp-tflite-micro:
     version: 1.3.3~1
   espressif/esp32-camera:
-    version: 2.0.15
+    git: https://github.com/RubenKelevra/espressif_esp32-camera.git
+    version: 2368d94-PR781
   espressif/mdns:
     version: 1.8.2
   espressif/esp_wifi_remote:
-- 
2.50.1
```

ctrochalakis, 2025-08-03:
Hello! I am also having this issue, using your patch above on top of esphome's dev branch, I 'm getting the following error:

```[17:46:53]cam_hal: NO-SOI - JPEG start marker missing

[17:46:56]***ERROR*** A stack overflow in task framebuffer_tas has been detected.

[17:46:56]Backtrace: 0x403768c9:0x3fca7240 0x4037bd61:0x3fca7260 0x4037c972:0x3fca7280 0x4037d7eb:0x3fca7300 0x4037ca78:0x3fca7320 0x4037ca6e:0x00000000 |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x403768c9: panic_abort at /home/yatiohi/.platformio/packages/framework-espidf/components/esp_system/panic.c:469
WARNING Decoded 0x4037bd61: esp_system_abort at /home/yatiohi/.platformio/packages/framework-espidf/components/esp_system/port/esp_system_chip.c:87
WARNING Decoded 0x4037c972: vApplicationStackOverflowHook at /home/yatiohi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
WARNING Decoded 0x4037d7eb: vTaskSwitchContext at /home/yatiohi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/tasks.c:3689 (discriminator 7)
WARNING Decoded 0x4037ca78: _frxt_dispatch at /home/yatiohi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
WARNING Decoded 0x4037ca6e: _frxt_int_exit at /home/yatiohi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

```

my config is

```yaml
# https://gr.mouser.com/new/m5stack/m5stack-u174-b-camera/
# https://docs.m5stack.com/en/unit/Unit-CAMS3%205MP
# https://www.reddit.com/r/Esphome/comments/1dt768b/esp32_camera_for_m5stack_cams3/
esp32:
  board: esp32-s3-devkitc-1 #This is the ESP32-S3-WROOM-1-N16R8 module
  flash_size: 16MB
  framework:
    #type: arduino
    type: esp-idf
    #version: "5.3.0"
    #platform_version: "6.9.0"
    #sdkconfig_options:
    # CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
     #CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
     #CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

psram:
  mode: octal
  speed: 80MHz

logger:
  level: VERBOSE

esphome:
  name: m5cams3cpy
  friendly_name: m5cams3cpy

esp32_camera:
  name: camera
  external_clock:
    pin: GPIO11
    frequency: 20MHz
  i2c_pins:
    sda: GPIO17
    scl: GPIO41
  data_pins: [GPIO6, GPIO15, GPIO16, GPIO7, GPIO5, GPIO10, GPIO4, GPIO13]
  vsync_pin: GPIO42
  href_pin: GPIO18
  pixel_clock_pin: GPIO12
  reset_pin: GPIO21
  resolution: 320x240 #800x600 #1600X1200
  jpeg_quality: 12
  #max_framerate: 10 fps
```

RubenKelevra, 2025-08-03:
Hey @ctrochalakis,

yeah, the default stack size seems to be too small, I already reduced the debug logging to reduce it. I'm still not sure what's using up all that stack in those cases.

I temporarily raised the stack size for this task, to circumvent this. Here's a snippet from my test config:

```yaml
esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  cpu_frequency: 240MHz
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CAMERA_TASK_STACK_SIZE: "8192" 
    advanced:
      enable_idf_experimental_features: yes

external_components:
  - source: github://pr#9771
    components: [esp32_camera]
    refresh: 0s

psram:
  mode: octal
  speed: 80MHz
  enable_ecc: true
```

I'm also testing on top of ESPHome 2025.7.4, to avoid running into not well tested code in the dev branch.

Fetching PR https://github.com/esphome/esphome/pull/9771 is unnecessary, unless you want to test out the PSRAM-DMA mode, which will currently just crash the device reliably ;)

turenkomv, 2025-08-06:
Unfortunately, I no longer have the UserDemo image — it was from a long time ago.

I can confirm that the camera still does not work with ESPHome 2025.7.5 and esp32-camera from PR #781. The following log is produced:

```
[13:30:56]cam_hal: NO-SOI - JPEG start marker missing - 100 additional misses
[13:30:57][D][esp-idf:000][framebuffer_tas]: W (26755) cam_hal: Failed to get frame: timeout
[13:30:57][W][esp32_camera:199]: Got invalid frame from camera!
```

patch:

```patch
From 66134b29996473aad750128b66485b6d5643aa20 Mon Sep 17 00:00:00 2001
From: Maksim Turenko <turenkomv@icloud.com>
Date: Wed, 6 Aug 2025 13:08:09 +0300
Subject: [PATCH] 2368d94-PR781

---
 esphome/components/esp32_camera/__init__.py      | 6 +++++-
 esphome/components/esp32_camera/esp32_camera.cpp | 2 +-
 esphome/idf_component.yml                        | 3 ++-
 3 files changed, 8 insertions(+), 3 deletions(-)

diff --git a/esphome/components/esp32_camera/__init__.py b/esphome/components/esp32_camera/__init__.py
index a99ec3408..2a4085c29 100644
--- a/esphome/components/esp32_camera/__init__.py
+++ b/esphome/components/esp32_camera/__init__.py
@@ -333,7 +333,11 @@ async def to_code(config):
     cg.add_define("USE_CAMERA")
 
     if CORE.using_esp_idf:
-        add_idf_component(name="espressif/esp32-camera", ref="2.0.15")
+        add_idf_component(
+            name="esp32-camera",
+            repo="https://github.com/RubenKelevra/espressif_esp32-camera.git",
+            ref="2368d94-PR781",
+        )
 
     for conf in config.get(CONF_ON_STREAM_START, []):
         trigger = cg.new_Pvariable(conf[CONF_TRIGGER_ID], var)
diff --git a/esphome/components/esp32_camera/esp32_camera.cpp b/esphome/components/esp32_camera/esp32_camera.cpp
index eadb8a440..95ae739a6 100644
--- a/esphome/components/esp32_camera/esp32_camera.cpp
+++ b/esphome/components/esp32_camera/esp32_camera.cpp
@@ -40,7 +40,7 @@ void ESP32Camera::setup() {
   this->framebuffer_return_queue_ = xQueueCreate(1, sizeof(camera_fb_t *));
   xTaskCreatePinnedToCore(&ESP32Camera::framebuffer_task,
                           "framebuffer_task",  // name
-                          1024,                // stack size
+                          8192,                // stack size
                           this,                // task pv params
                           1,                   // priority
                           nullptr,             // handle
diff --git a/esphome/idf_component.yml b/esphome/idf_component.yml
index c43b62268..186e97304 100644
--- a/esphome/idf_component.yml
+++ b/esphome/idf_component.yml
@@ -2,7 +2,8 @@ dependencies:
   espressif/esp-tflite-micro:
     version: 1.3.3~1
   espressif/esp32-camera:
-    version: 2.0.15
+    git: https://github.com/RubenKelevra/espressif_esp32-camera.git
+    version: 2368d94-PR781
   espressif/mdns:
     version: 1.8.2
   espressif/esp_wifi_remote:
-- 
2.49.0.windows.1
```

Without increasing the framebuffer_task stack size, I was getting the following error:
```
[13:27:21]***ERROR*** A stack overflow in task framebuffer_tas has been detected.
```

closed 2025-10-26 by github-actions[bot]