espressif/esp32-camera #775: PSRAM DMA Mode does not work with JPEGs

RubenKelevra, 2025-07-21:
> > Have you recently tested this mode?
> 
> success depends on different things. It does not work well for JPEG, but works fairly well for other modes. That is if XCLK is 16MHz... other sample clocks do not work well. You can see issues if PSRAM is otherwise heavily used too. Overall experimental feature

Yeah I think I found the issue why it's not working for JPEGs:

The test for the start of the byte marker for JPEGs does not work as far as I can tell, because the length field is not updated. So previous to #760 it would just try to read random bytes from the memory.

Now it's just returning because there's nothing to read, because the length is zero.

What needs to be done is, that if we get the first EOF for a completed DMA copy that we copy the first section from the PSRAM back into a temporary buffer in the SRAM, and give this temporary buffer to the SOI checker, so it can verify that there's a valid JPEG start.

I'll provide an PR to do this as soon as I get time for that.

_Originally posted by @RubenKelevra in https://github.com/espressif/esp32-camera/issues/774#issuecomment-3095973580_

jameszah, 2025-07-31:
Is this the problem that presents as follows (100 lines maybe), and then starts working normally ... but comes and goes ???

```
21:14:24.293 -> E (4204) cam_hal: DMA overflow
21:14:24.387 -> E (4292) cam_hal: DMA overflow
21:14:24.387 -> E (4332) cam_hal: DMA overflow
...
21:14:28.749 -> Camera Capture Failed
```

This is using old code with VGA frames at 25 fps, after 3.2.0 -> 3.3.0 upgrade, on esp32-cam ai-thinker.

RubenKelevra, 2025-07-31:
@jameszah not sure what this version number is referring to, as this library is currently at version 2.1.0.

If you use 2.1.0, PSRAM DMA is only active if you select 16 MHz as XCLK on an ESP32S2 or ESP32S3.

If you use the master of this library, it's only active if you set CONFIG_CAMERA_PSRAM_DMA to 1 on an ESP32S2 or ESP32S3.

The module you're describing sounds like an ESP32S, which does not support psram_mode.

RubenKelevra, 2025-07-31:
@jameszah the current heuristic would make a memory slot for your images, 61440 bytes (if you're using VGA). Can you please try to set a higher number, like make it 256 kByte or something like this and report back if the issue goes away?

me-no-dev, 2025-07-31:
this is usually a problem only on higher resolutions, so maybe we could just divide `cam_obj->width * cam_obj->height` by `4` for them and allow larger sizes

RubenKelevra, 2025-07-31:
> If you are trying to say that the DMA could write more data than the FB size, then that is just not true

I've looked at the code, and I see what you mean.

Yeah, there's a check for the size of the image. But this only works if we're not in `psram_mode`. On one hand because the check is guarded by `if(!cam_obj->psram_mode){` ([Link to source](https://github.com/espressif/esp32-camera/blob/2368d94f11d0cb6d8ec7b153fa4390f8cd1f83cb/driver/cam_hal.c#L201)) and on the other hand because it depends on us doing the copy with the CPU.

In `psram_mode` we are just called after the copy and thus need to check if there's still one buffer left to write into for the next camera DMA copy operation. And that's what I'm doing in PR https://github.com/espressif/esp32-camera/pull/772.

However, there's are two issues with this approach:

- We need one memory buffer *more* space for the images, which I did not account for.
- We currently do the check even if not in psram_mode, which is unnecessary and thus reduces our buffer size by 1 dma buffer unnecessarily, if we're not in psram_mode.

I'll provide a fix.

RubenKelevra, 2025-07-31:
Hey @jameszah,

thanks for providing more info on that matter. But I've already found the issue and provided a fix.

In short terms, a new check which was meant just for the `psram_mode`, which prevents buffer overflows, was inadvertently also running if the psram_mode is not active.

I also failed to account for the fact, that it will reduce the size for the frame buffer slots by one memory copy operation length. Therefore, you saw a reduction in maximum possible size for images.

I've moved the check to only be active in `psram_mode` and increased the frame buffer size for the `psram_mode` to account for the reduction in size caused by the different check logic, if the `psram_mode` is active.

jameszah, 2025-09-24:
I agree - works fine on IDE 2.

RubenKelevra, 2026-03-28:
@romainvandooren thanks for the detailed information, I'll have a look later today.