espressif/esp32-camera #797: Setting Set CONFIG_LOG_DEFAULT_LEVEL_NONE=y but still got some message from console print

AxelLin, 2025-08-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Test with ESP-IDF v5.4 branch.
Set CONFIG_LOG_DEFAULT_LEVEL_NONE=y , but I still observed the below messages:

cam_hal: FB-SIZE: 832000 != 960000
cam_hal: FB-SIZE: 832000 != 960000
cam_hal: FB-SIZE: 848000 != 960000
cam_hal: FB-SIZE: 848000 != 960000
..

This seems wrong.

### Actual behavior (suspected bug)

It should not print anything if CONFIG_LOG_DEFAULT_LEVEL_NONE=y.

### Error logs or terminal output

```shell
cam_hal: FB-SIZE: 832000 != 960000
cam_hal: FB-SIZE: 848000 != 960000
cam_hal: FB-SIZE: 848000 != 960000
```

### Steps to reproduce the behavior

Just normal build. 

### Project release version

v2.1.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

Bash

### Additional context

This issue seems to be introduced by c1c0da435eac3bb7037342991d88ec3ec1f3d4b8

me-no-dev, 2025-08-22:
@AxelLin can you try this: https://github.com/espressif/esp32-camera/pull/798

AxelLin, 2025-08-22:
> [@AxelLin](https://github.com/AxelLin) can you try this: [#798](https://github.com/espressif/esp32-camera/pull/798)

@me-no-dev 
Thank you for the quick fix.
I just tested it, the fix looks good.

me-no-dev, 2025-08-22:
thanks for confirmation :) merged